logisim-evolution/logisim-evolution #505: macOS code signature resources missing?

meow10811, 2021-02-17:
I've recently had to use Logisim for my university class, but I've run into a couple problems trying to use the .dmg installer - the same "This application is damaged and can't be opened" that came up in #475, #476, and #499. I've looked into it, and found that macOS's Gatekeeper is restricting the app because the "code has no resources but signature indicates they must be present" (according to `codesign`/`spctl`). I'll try to build the .app locally and see if it has the same problem, but it seems like something in the package might have to be re-signed.

meow10811, 2021-02-17:
I built the app locally and it obviously wasn't signed, so I was unable to reproduce the issue. I've done a bit more research, and it looks like `jpackage`'s default settings seems to sign the application incorrectly (according to [this StackOverflow post](https://stackoverflow.com/questions/60953329/code-signing-notarization-using-jpackage-utility-isnt-working-on-macos)). I'm not sure who to raise this issue with, but would it be possible for someone to re-sign the `.dmg` properly so that Mac users don't have to compromise their security settings or use the universal `.jar`. still?

davidhutchens, 2021-02-17:
If you build it yourself, I'd recommend using the develop branch. It has some fixes for the Mac.

However, the cause of the failure with Gatekeeper is that the app isn't signed at all. The Mac tags it as quarantined when you download the dmg. It copies that tag to the app when you drag it out of the dmg. You can see this from Terminal with "xattr -l Logisim-evolution.dmg". The solution listed in the issues you tag:
     xattr -cr Logisim-evolution.app
simply removes the quarantine attribute (c=clear, r=recursive) and the app runs as expected.

What we would need to do to fix the issue is actually sign it with a Mac developer certificate when it is built.

meow10811, 2021-02-17:
Hm, I see - I tried uploading my built `.dmg` to the Internet and redownloading it, and I do indeed get a permissions error. The error I get for my own disk image is different, though - it simply states that the app's developer can't be verified, and macOS still lets me open it by left-clicking. In the version I can download from GitHub, it doesn't even let me override the security settings.

davidhutchens, 2021-02-17:
That is curious. Can you give me a link to the version you uploaded? I want to see what it does for me.

meow10811, 2021-02-17:
[Here's the link to my version](https://mega.nz/file/AVQm0R4T#YuCkxyLpEzreOTjVVYcbIoFRadQ0XgJ1Gzbsh4gDYVQ) - this is straight out of build/distributions in my local copy of the repo.

davidhutchens, 2021-02-17:
Thank you. I can download your dmg, drag out the app, and open it with left-click and select open.

I just uploaded a copy to my web pages and then downloaded it. The app does not open with left-clicking and selecting open.

So what is the difference?

 I'm running MacOS 10.14.6 with all security updates installed. I compiled using Java 15 2020-09-15. Build 15+36.

meow10811, 2021-02-17:
I have no clue what the difference could be, but I can tell you that this copy (which I just rebuilt), as well as the first copy which I gave you were both compiled on my macOS 10.15.7 system with all security updates installed using OpenJDK 15.0.1 2020-10-20 Build 15.0.1+9, from the latest master branch commit. Maybe if you were compiling off of the develop branch?

maehne, 2021-02-17:
@meow10811  and @davidhutchens: Thanks a lot for looking into this issue! The "This application is damaged and can't be opened" is really annoying, as it creates doubts that the applications integrity was compromised. As a first step, we should try to ensure that we get again the old behaviour, where macOS was reporting that the app's developer can't be verified.

@meow10811: Could you maybe also provide a built from the HEAD of the `develop` branch?

meow10811, 2021-02-17:
@maehne Absolutely - [here's a build](https://mega.nz/file/MAoXGYyL#FezyYe4JOM5OAl3ZlxX9Epj2lpDo9r28Vn9swfiokqY) from the HEAD of the `develop` branch. For some reason, the tasks under the `macappstore` folder in Gradle don't appear in this branch, so I had to build using `jpackage` instead of `createDmg`. I went back and tested building from `master` using `jpackage` ([here](https://mega.nz/file/4FplXIjZ#YfmaphwPuFbavzqcOQ2JnlgPqnsG1i7gEUfIIc6HIVE)), and I still get the same result.

This is getting more and more confusing - every single build I've created/uploaded hasn't had the signature issue, but everyone else's has, for some reason?

davidhutchens, 2021-02-17:
I have been looking at this. You are exactly right as to the difference. Building with createDmg from earlier builds works to get an usable unsigned app in a dmg for at least 10.14 and 10.15.

Thank you @meow10811 for the helpful clues.

Since you are getting a usable dmg from the jpackage build, it is either because you are using MacOS 10.15 or (my bet) they fixed something in the jdk builds between us. I'll upgrade my jdk to see what that does for me. I'll report back later.

davidhutchens, 2021-02-17:
I lose the bet. I upgraded to OpenJDK 15.0.2. The generated dmg from HEAD of develop doesn't work when downloaded from my web page.

That leaves me with MacOS 10.15 working to build the dmg and MacOS 10.14 not. Perhaps due to changes in newer versions of Xcode?

@maehne, do you have a MacOS 10.15 machine you could test this on?

meow10811, 2021-02-18:
I come with more updates! I got my hands on a macOS 10.14.6 Mojave computer, installed the same version of OpenJDK 15 from Homebrew, and downloaded and built the project using IntelliJ IDEA CE 2020.3 - and it still functions properly. That was as controlled of a study as I could make - there was no major difference between my previous builds and these builds (which you can find [here](https://mega.nz/file/ABhHUYYB#ltc9PMPrgMIp8pEx6NkQDW4mfrqvU6WuWPICq9mXYCM) for the `jpackage` version and [here](https://mega.nz/file/4M5BSA4a#bA0eGgxhv7d9ZLifsf6ID7v2KaFu4dl49cphVD0ZiRY) for the `createDmg` version).

Maybe the behavior depends on what site is hosting the files?

davidhutchens, 2021-02-18:
That is more info... But it isn't the site. I tried uploading your working jpackage dmg to my web page and then downloaded it. It works. So it isn't the site. But that sure does leave us wondering what the difference could be.

I use the command line to build it. But that seems unlikely to be the difference. I'm puzzled.

davidhutchens, 2021-02-19:
I have Xcode version 11.3.1. What version were you running on the 10.14.6 computer?

davidhutchens, 2021-02-19:
@BFH-ktt1, could you please give us the specific versions of MacOS, Xcode, and OpenJDK you used to build the 3.4.2 Mac dmg? It reports it is damaged (as does 3.4.1) on my machine. Thanks.

BFH-ktt1, 2021-02-19:
@davidhutchens , I build it with an "old" Macbook air, and after I build the dmg it started updating, so I'm not sure that the info after the update is correct. Starting from next week I'll have a mac mini based on an M1-chip (model late 2020), so probably than the problem will (hopefully) be solved...

meow10811, 2021-02-19:
@davidhutchens I faced a similar update problem with Xcode but it reports as 11.3.1 on the Mojave 10.14.6 computer. The other computer that I built on runs Xcode 12.3.

maehne, 2021-02-20:
@meow10811: Thanks for your investigations and different DMG builds for testing!

I got around today to build Logisim-evolution from current HEAD of the `develop` branch using `./gradlew jpackage` on my MacBook Pro, which is still stuck macOS High Sierra 10.13.6 with OpenJDK 15.0.2 and Xcode 10.1. The installing the app bundle from the locally created DMG works. Once, I upload the DMG to a web site and download it again, it gets reported as damaged. The same happens if I build just the app bundle with `jpackage`, put it into a ZIP archive, upload it to a web server, download it, and install it. However, if I run the following command on the app bundle:

```
codesign --remove-signature Logisim-evolution.app
```
I can put it into a ZIP archive, which I can download from a web site and the unpacked app bundle will just report that it is from an unverified developer. This works too if I create a DMG by calling `jpackage` again on the app bundle:

```
jpackage --type dmg --app-image Logisim-evolution.app --app-version 3.4.2 --name Logisim-evolution
```

So, we need to modify the `gradle` build script to:

1. call `jpackage` to build just the app bundle
2. remove the signature with `codesign`
3. call `jpackage` again to build the DMG.

This two-step approach is also [described](https://docs.oracle.com/en/java/javase/14/jpackage/image-and-runtime-modifications.html#GUID-70F25F63-DE4C-4DC0-83D5-76EB5C55BEF0) in the `jpackage` documentation. Unfortunately, I won't get to do the necessary modifications myself in the coming days due to other duties.

davidhutchens, 2021-02-21:
Thanks @maehne.  I'll look at it this weekend.

davidhutchens, 2021-02-21:
It looks even easier. I was already fixing the Info.plist with a post app creation script (called from inside jpackage). My tests suggest if I just add the codesign remove command to the script, I get an app that does not appear to be broken.

@maehne and @meow10811, could you download the Logisim-evolution.dmg from http://cs.millersville.edu/~dhutchens/Logisim/ and see if it works for you?

If it looks good, I'll prepare a PR.

davidhutchens, 2021-02-21:
I also now have a version that works as @maehne suggested. The advantage of that is that the patch up script is more a more obvious part of the build in the build.gradle.kts file. The disadvantage is that it takes 25-30 more lines of code in that file to run the three processes (create app, fix it, create dmg)

@maehne and @BFH-ktt1, do you have a preference?

meow10811, 2021-02-21:
@davidhutchens I can confirm the version you linked doesn't report as damaged. If I'm reading what you diagnosed right, was it just one line to force-unsign the application during the build process? I'm still trying to figure out why I couldn't build a damaged version of the application.

davidhutchens, 2021-02-22:
@meow10811, I am still puzzled by your ability to build an undamaged app. I upgraded to 10.15.7 to see if that made any difference on my end. No change. Well, it is harder to authorize the unsigned app since I need to do so in System Preferences / Security after attempting to open it with right-click.

@maehne, do we need some kind of "How to install on the Mac" documentation? If so, where should it be placed?

R3dst0ne, 2021-11-06:
@davidhutchens
What is the status on this one?
If I'm not mistaken, #1271 changed view things regarding this issue.

davidhutchens, 2021-11-06:
Thank you @R3dst0ne. I thought this issue was closed in February. It should be. Yes, #1271 should help also.

I can't close it. @BFH-ktt1, would you do so please?

closed 2021-11-07 by BFH-ktt1