logisim-evolution/logisim-evolution #507: Allow 64 bit width Test Vector file

simoc, 2021-02-19:
I can create circuits with 64 Data Bits in logisim evolution, but I get the following message in an error dialog when I open a Test Vector file with 64 bits:

Test vector in "test64.txt" could not be parsed. Detail: Test Vector header format error: bad width: InA[64].

simoc, 2021-02-19:
One problem is the `if (w < 1 || w > 32)` comparison in `TestVector.java` that directly leads to this error message.

After changing the `32` here to `64`, there are further problems parsing 64 bit values in `Value.java` and I get the following error message:

 Test Vector data format error: too many bits in "0x00000000000000f1".

I have the same problem if I define the values in binary.

maehne, 2021-02-19:
You may want to check out PRs #327 and #394, which implemented the support for larger busses than 32 bit. The changes made there should be similar to the ones, which are now needed to adapt the Test Vector code to these larger bus widths.

simoc, 2021-03-13:
Fixed by #508. I am closing this issue manually.

PR #508 (2021-03-04): Allow 64 bit Test Vector file
Allow 64 bit width in Test Vector file header.

Removed unnecessary "too many bits" check whilst reading digits of Test Vector file values. This test does not work for 64 bits (`value >> 64` always returns the same 64 bit `value`). The test at the end of the loop after parsing all digits is already adequate for testing "too many bits".

Fixes #507
closed 2021-03-13 by simoc