logisim-evolution/logisim-evolution #509: Java exceptions when adding Slider to empty circuit

simoc, 2021-02-21:
After starting logisim evolution and adding a Slider to my new circuit, many Java exceptions are reported in the terminal window and logisim is no long usable.

The first Java exceptions are:

    java.lang.ClassCastException: class java.lang.Integer cannot be cast to class java.lang.Long (java.lang.Integer and java.lang.Long are in module java.base of loader 'bootstrap')
	at com.cburch.logisim.std.io.extra.Slider.propagate(Slider.java:301)
	at com.cburch.logisim.instance.InstanceComponent.propagate(InstanceComponent.java:426)
	at com.cburch.logisim.circuit.CircuitState.processDirtyComponents(CircuitState.java:383)
	at com.cburch.logisim.circuit.Propagator.clearDirtyComponents(Propagator.java:226)
	at com.cburch.logisim.circuit.Propagator.propagate(Propagator.java:291)
	at com.cburch.logisim.circuit.Simulator$PropagationManager.run(Simulator.java:128)
    Exception in thread "AWT-EventQueue-0" java.lang.ClassCastException: class java.lang.Integer cannot be cast to class java.lang.Long (java.lang.Integer and java.lang.Long are in module java.base of loader 'bootstrap')
	at com.cburch.logisim.std.io.extra.Slider.getValueState(Slider.java:179)
	at com.cburch.logisim.std.io.extra.Slider.paintInstance(Slider.java:273)
	at com.cburch.logisim.instance.InstanceComponent.draw(InstanceComponent.java:255)
	at com.cburch.logisim.circuit.Circuit.draw(Circuit.java:549)
	at com.cburch.logisim.gui.main.CanvasPainter.drawWithUserState(CanvasPainter.java:167)
	at com.cburch.logisim.gui.main.CanvasPainter.paintContents(CanvasPainter.java:227)
	at com.cburch.logisim.gui.main.Canvas.paintComponent(Canvas.java:532)
	at java.desktop/javax.swing.JComponent.paint(JComponent.java:1074)

I am using the `develop` git Branch.

zdimension, 2021-03-12:
If this issue was fixed by #510, it should be closed, I think

simoc, 2021-03-13:
Yes, this issue is fixed by #510. I am closing this issue manually.

Pull request #510 has the "Fixes" reference in the description to automatically close this issue after merging. However, this only works for merging to the default branch for the logisim-evolution repository (master). Logisim-evolution pull requests are merged to branch develop instead.

PR #510 (2021-03-04): Initialize Slider value attribute to 64 bit zero value
The Slider value attribute is used in class `com.cburch.logisim.std.io.extra.Slider` as a 64 bit value, so must be initialized as a 64 bit value with type `java.lang.Long` instead of type `java.lang.Integer`, to avoid java casting exceptions.

Fixes #509
closed 2021-03-13 by simoc