logisim-evolution/logisim-evolution #515: I want to make a truth table that needs 4-bit width How do I write it ??

nandiabhineel, 2021-03-05:
As a part of our lab experiment we had to construct an ALU in logisim.
For validating the experiment I wanted to make use of test vector ,
but I am confused as to how can I parse or make a truth table in .txt file such that some of it's fields should be 4 bits long.
I tried by watching out some reference but they all were for single bit fields in Truth Table. 
Please tell me how to make or format my .txt file to incorporate 4bit width values like 1011, 1111 , e.t.c

R3dst0ne, 2021-07-03:
Example from the documentation(`Guide to Being a Logisim User`>`Command-line verification`>`Text Vectors`):
>Here is an example of a test vector file:
#test vector for adder 4bit 
A[4] B[4] O[4] C_in C_out
0000 0000 0000 0 0
0000 0000 0001 1 0
0xf 0xf 0000 1 0
0xf 0xf 0001 1 1

I hope this helps.

MarcinOrlowski, 2021-07-03:
@maehne This one should be moved to Discussions I think.

closed 2021-07-03 by maehne