logisim-evolution/logisim-evolution #516: Add documentation explaining Assembly window

simoc, 2021-03-06:
The online help under "Guide to Being a Logisim User" -> "Menu reference" -> "The Simulate menu" contains only

**Assembly viewer**
not documented yet

simoc, 2021-03-06:
After experimenting with this window and reading the source code in `AssemblyWindow.java`, I think the documentation should be:

The Assembly window displays the value of a register in the current circuit, together with the assembly language instruction at the address in the register.

Select the register for display and load a text file containing the assembly language instructions.

The filename must have a `.lss` suffix. An example text file is:

```
    0: LOD A,[1005h]
    3: ADD A,[1001h]
    6: STO [1005h],A

    9: LOD A,[1004h]
    C: ADC A,[1000h]
    F: STO [1004h],A

   12: LOD A,[1003h]
   15: ADD A,[001Eh]
   18: STO [1003h],A

   1B: JNZ 0000h

   1E: FFh

 1000: 00h, A7h
 1002: 00h, 1Ch
 1004: 00h, 00h
```

Instruction addresses are in hexadecimal with at least one leading space and without leading zeroes, followed by a colon and the assembly language instruction.

Pressing `F2` in the Assembly window advances the clock by two ticks.

simoc, 2021-03-06:
Attaching screenshot of Assembly window using the example text file from my comment above.

The register `PC` currently contains the value `0x15`. This line is highlighted in the Assembly window.

[image]

maehne, 2021-03-13:
I guess this issue can be closed due to merged PR #518.

PR #518 (2021-03-13): Added English language Assembly viewer documentation
Added HTML online help and screenshot for Simulate -> Assembly viewer menu option.

Copied this documentation to directories for all other languages, following the organisation of the previous menu option Simulate  -> Test Vector.

As an extra step, I could translate this new documentation to the other languages using [deepl](https://www.deepl.com/translator), if wanted. Translation does not seem to be handled consistently for other documentation. Some of it has been translated, some left in English, so I am not sure what is best.

Fixes #51 [... cut]