logisim-evolution/logisim-evolution #521: travis-ci.org is shutting down in several weeks

maehne, 2021-03-13:
[travis-ci.org](https://travis-ci.org/) is shutting down in several weeks, with all accounts migrating to [travis-ci.com](https://travis-ci.com/). Details about what this means for this project are yet missing. They offer a [free account](https://travis-ci.com/plans) also on travis-ci.com, but with certain restrictions. So, it may be better to switch to [GitHub Actions](https://github.com/features/actions) or another solution.

**Suggestions and help in this area are welcome, as we are not yet very familiar with CI/CD flows!**

zdimension, 2021-05-08:
16d479cee999f715c353271dfbf8ad6fe1af3e69 adds GitHub Actions support.

zdimension, 2021-05-10:
I've been unable to find the exact shut down date of travis-ci.org (it's been saying "in several weeks" for six months) but I think it's a safe guess to assume that it can close at any moment now. If @BFH-ktt1 is okay with it, I can remove the travis-ci files in #560

BFH-ktt1, 2021-05-10:
For me it is okay to remove travis support as it is not working anyways already for a while.