logisim-evolution/logisim-evolution #522: Random Generator generates only 32 bit values, not 64

simoc, 2021-03-13:
A Random Generator component with Data Bits 64 generated the following sequence of values for me:

    ffffffffdd309afb
    ffffffffc92dcf63
    000000006c38104d
    ffffffff997c3694
    00000000563d242a
    ffffffffccc47aa3

These look like 32 bit random values, not 64 bit random values.

I am using the `develop` git Branch.

simoc, 2021-03-13:
The online help says that the Random Generator component values follow the sequence `r1 = (25214903917 * r0 + 11) modulo 2^48`.

This means that the random values should have 48 bits of randomness.

However, the Java class `com.cburch.logisim.std.memory.Random` uses a 32 bit `int` type in the calculations, so some of the randomness is lost.

As an alternative, I can use two 32 bit Random Generator components and a Splitter to generate a 64 bit random value.

maehne, 2021-03-14:
Thanks @simoc for reporting this issue. We will need to replace the PRNG implementation to generate 64-bit pseudo-random values. We should use a proven algorithm. E.g., since Java 8, there is [`java.util.SplittableRandom`](https://docs.oracle.com/javase/8/docs/api/java/util/SplittableRandom.html), which implements [Splitmix64](https://rosettacode.org/wiki/Pseudo-random_numbers/Splitmix64). There are also many [alternative PRNG algorithms](https://prng.di.unimi.it). There is [Apache Commons RNG](https://commons.apache.org/proper/commons-rng/userguide/rng.html), which provides Java implementations for some of them.

In light of an efficient implementation on FPGA, an algorithm avoiding multiplications and relying instead fully on XOR, shift, rotate, and addition operations may be preferred. [xoshiro256++](https://prng.di.unimi.it) seems to be a good candidate. There's a [Java implementation](https://dsiutils.di.unimi.it/docs/it/unimi/dsi/util/XoShiRo256PlusPlusRandom.html), but implementing it ourselves along the lines of [this](https://unascribed.com/b/2020-01-31-i-ported-xoshiro256-to-java-so-you-dont-have-to.html) would also not be a big effort. In any case, we will have to implement VHDL/Verilog versions of it.