logisim-evolution/logisim-evolution #524: UI scaling after opening of a file dialog

NikLeberg, 2021-03-16:
Hello there.

After I've updated to the latest version (3.4.2, I've installed it with the .msi windows installer) I encounter a strange bug.
I can open and edit .circ files with no problem.
But when I open any sort of "file-dialog" like:
- File -> Open
- File -> Merge
- Project -> load library -> Logism-evolution library
- Simulate -> Test Vector
at the moment the window opens the main logism window goes all wonky: enlarged scollbars and enlarged font.
But it's still completely usable and the simulation works fine.

After I close logism and reopen it, it's back to normal. (Untill I click one of the menitioned dialogs again..)

Some screenshots:

before:
[image]

after opening the dialog:
[image]

logism afterwards:
[image]

Maybe it has to do with the fact, that my machine has a 4k display. Many programs seem to dislike that (LTspice as example). But usually the reverse effect happens: all buttons or texts are super small...

Windows: 10.0.19042.867 (with UI-Scaling 250%)
AdoptOpenJDK: jdk-15.0.2.7-hotspot

Maybe related to #444 ?

Thanks,
Nik

NikLeberg, 2021-03-16:
Just now I also discovered, thats this bug is also triggered if I switch to tablet mode in windows.

maehne, 2021-03-22:
Thanks for reporting this bug. This is probably due to different parties (i.e., Windows, the JDK, and Logisim-evolution itself) trying to scale the GUI widgets to the right size. You have set the Look and Feel to the native "Windows" style, which is partly drawn by Windows and partly by Java. To work around the problem, please try to switch in "File -> Preferences... -> Window" the "Look and Feel" to "Metal", which is completely drawn by Java. This seems to yield a homogeneous scaling of all GUI elements. Another thing, which you can try, is to set in the same window preferences pane the scaling factor to 1.0.

@BFH-ktt1 implemented the scaling feature in Logisim-evolution to make it usable on high-resolution screens, when there was no support for it yet in the JDK and also Windows was not yet so good in scaling. We may need to consider to remove it again, now that there is native support for scaling on the different OS platforms.

NikLeberg, 2021-03-23:
Thanks for the suggestion. I guess I forgot that I had made that change to the look and feel. Changed to Metal and now it works flawlessly.
I also played around a bit with the different styles:
- Metal - works fine
- Windows - mentioned bugs
- Windows Classic - bug only triggers when entering tablet mode. The other "bug-paths" e.g. `File -> Open...` do not trigger it.
- the others - work fine, but I didn't test them all. I guess those are all handled by the jdk and windows can't screw anything up.

Maybe this helps to track down the source of the scaling issue.

Feel free to close this issue.

maehne, 2021-03-23:
Thanks @NikLeberg for your feedback and additional tests with other look and feels! Be aware that some recently added look and feels based on the themeable Darklaf look and feel seem to not work on Windows (see issue #535) and may require to delete a key using the Windows Registry Editor to make Logisim work again.

I am glad that the Metal look and feel does work correctly on high-resolution screens as a workaround. Still, it would be good to find a fix for these issues. As our resources are limited, this issue won't be high on our priority list, but we appreciate any external contributions to this or other issues!

hewzhew, 2026-08-29:
I rechecked this against current `main`. The original report used v3.4.2 with JDK 15, Windows L&F, and 250% display scaling. Current builds require Java 21+ and default to FlatLaf.

On JDK 25, an automated probe did not show any global UI metric changes while initializing `JFileChooser` under Windows L&F, including forced 250% Java UI scaling. I do not have a real 4K/250% Windows setup, so this is not an exact retest.

Since the reporter previously confirmed the workaround and was fine with closing, would it be reasonable to close this as obsolete unless it can still be reproduced on a current build?

maehne, 2026-09-01:
I am closing this as resolved due to the use of Java 21+. Please reopen if the issue persists.

closed 2026-09-01 by maehne