logisim-evolution/logisim-evolution #535: Problems with recently added "Darklaf" Swing Look and Feels

maehne, 2021-03-23:
I stumbled on the new themeable "Darklaf" Swing Look and Feels introduced by commit 2cabc67a5 by @zdimension. They work on macOS, but I observed some visual issues. Especially, the icons for some components are not anymore shown in the Library pane on the left, e.g., for D Flip-Flop, T Flip-Flop, J-K Flip-Flop, S-R Flip Flop in the Memory folder, but also other components are affected. It doesn't seem to be only an issue of bad contrast.

Worse is the behaviour on Windows 10: The main window won't open after restarting Logisim-evolution. To reset the Look and Feel to "Metal", I had to open the Windows Registry Editor to delete the key `/Look/And/Feel` in `Computer\HKEY_CURRENT_USER\SOFTWARE\JavaSoft\Prefs\com\cburch\logisim`.

@zdimension: On which platforms did you test the new look and feels? It would be nice if they were working reliably. Maybe, you can find a fix? If not, it is probably better to revert the commit 2cabc67a5.

zdimension, 2021-03-25:
Really sorry about that, I tested the new themes on Windows 10 and Ubuntu 20.04 and didn't encounter any of those problems. The icons are displayed correctly for me:
[image]
Admittedly I did not test on macOS since I don't have a Mac computer nor a VM with me.

The second bug, the one where the main window won't open, is concerning. Did you get any error message? Also, did you try starting with a debugger to see if something is crashing internally?

maehne, 2021-05-19:
I believe issue #566 is the same bug. @Moi78 got the following error:

```
Exception in thread "main" java.lang.IllegalAccessError: class com.github.weisj.darklaf.ui.optionpane.DarkOptionPaneUI (in unnamed module @0x1efbd816) cannot access class sun.swing.DefaultLookup (in module java.desktop) because module java.desktop does not export sun.swing to unnamed module @0x1efbd816
        at com.github.weisj.darklaf.ui.optionpane.DarkOptionPaneUI.createButtonArea(DarkOptionPaneUI.java:42)
        at java.desktop/javax.swing.plaf.basic.BasicOptionPaneUI.installComponents(BasicOptionPaneUI.java:205)
        at java.desktop/javax.swing.plaf.basic.BasicOptionPaneUI.installUI(BasicOptionPaneUI.java:159)
        at java.desktop/javax.swing.JComponent.setUI(JComponent.java:685)
        at java.desktop/javax.swing.JOptionPane.setUI(JOptionPane.java:1851)
        at java.desktop/javax.swing.JOptionPane.updateUI(JOptionPane.java:1873)
        at java.desktop/javax.swing.JOptionPane.<init>(JOptionPane.java:1838)
        at java.desktop/javax.swing.ProgressMonitor$ProgressOptionPane.<init>(ProgressMonitor.java:161)
        at java.desktop/javax.swing.ProgressMonitor.setProgress(ProgressMonitor.java:291)
        at com.cburch.logisim.gui.start.Startup.autoUpdate(Startup.java:581)
        at com.cburch.logisim.Main.main(Main.java:74)
```

If Logisim-evolution cannot start due to corrupt preference settings, we should maybe consider adding a startup switch, which would reset all preferences by deleting the locally saved preferences.

maehne, 2021-06-11:
Closing this due to merging of PR #619. The visual issues are tracked in issue #641.

closed 2021-06-11 by maehne