logisim-evolution/logisim-evolution #537: bidirectional pins

Barnaboss, 2021-03-24:
when designing a little cpu for funsies i wanted to tidy up my designe a little and therefore wanted to enclose every module that communicates with the bus into a subcircuit. Unfortunately it seems that it is not possible to configure pins to be bidirectional, so i want to suggest such an option, i think it makes alot of sence when designing components that communicate with a bus.

maehne, 2021-03-26:
Indeed, it would be nice to have support for inout ports in Logisim-evolution and it was suggested before (cf. to issue #253). Unfortunately, the simulation engine of Logisim-evolution is not able to handle inout ports and would need a rewrite, for which we are lacking the resources (see [comment](https://github.com/reds-heig/logisim-evolution/issues/253#issuecomment-484696612) by @kevinawalsh for details).

@BFH-ktt1: I propose that we close this issue as a duplicate of issue #253.

TChapman500, 2021-03-27:
I have an idea for a workaround:  If the pin receives input from within the subcircuit, set the internal direction to output, storing which bits are being driven to which values for that direction.  If the input comes from outside the circuit, set the internal direction to output, storing which bits are being driven to which values for that direction.  If the driving bits of the two directions contradict each other, propagate an error on both sides of the pin.

maehne, 2021-03-30:
@TChapman500: Thanks for suggesting a workaround to implement this feature! Contrary to @kevinawalsh and @BFH-ktt1), I am not familiar enough with the implementation of the simulation engine to judge it. Our development resources are currently limited and in the short term focused on fixing bugs, backporting features from other forks, improving support for synthesising designs on FPGA boards, improving documentation/localisation. However, we very welcome pull requests that implement new features and try to support new developers through code reviews and feedback to questions on a best effort base. Though, please continue discussing your idea in issue #253 and not this duplicate issue.

MarcinOrlowski, 2021-06-08:
While that's not really the direct solution but as (hopefully :) temporary workaround for someone in needs - the [Digital](https://github.com/hneemann/Digital) tool mentioned in README seems to support bidirectional ports. Haven't checked it yet, but at its library features 74245 IC, so...

closed 2021-07-14 by BFH-ktt1