logisim-evolution/logisim-evolution #542: Base Library Missing

Wynadorn, 2021-04-07:
In the original Logisim a category called Base is available.
[image]

This category is necessary for: **Project -> Options -> Toolbar** and **Project -> Options -> Mouse**
Without this section you're unable to bind these tools to keybinds or put them on the toolbar.

BFH-ktt1, 2021-04-07:
This is intentional. The "Base" tools are default in the Toolbar.
Closing the issue.

Wynadorn, 2021-04-07:
Sorry, I should have been more detailed about my issue.

I'd like to assign a hot key to these tool buttons. This uses to be able underneath the Mouse tab however because the base tab has (intentionally) been removed I'm no longer able to.

BFH-ktt1, 2021-04-07:
This makes sense, I will look into the details in due time, however, it may take a while

BFH-ktt1, 2021-04-25:
This should be fixed in my latest push to develop. Please reopen the issue if it is not resolved. (note it is not yet available in 3.4.5)

PR #556 (2021-04-25): Base again available for template and mouse
fixes #542
closed 2021-04-25 by BFH-ktt1