logisim-evolution/logisim-evolution #552: Project will not simulate unless all probe points are deleted

CuriousMarc, 2021-04-25:
Here is the same project, one with probe points, one without probe points. The one with probe points will refuse to run. The clock will not start. If you delete all the probe points one by one, after you delete the last one, it will run again. If you add them back in one by one, it will run too. But if you save the project with the monitors and reload it will stop to run. Removing all the probe points will get it to run again. Using 3.4.4. Thanks!
[Probe Points Bug.zip](https://github.com/reds-heig/logisim-evolution/files/6371215/Probe.Points.Bug.zip)

BFH-ktt1, 2021-04-25:
Thank you very much for your bug report. This has been fixed in my latest push in the development branch. Closing the issue for now, please reopen if it does not work for you.

closed 2021-04-25 by BFH-ktt1

CuriousMarc, 2021-04-25:
Wow that was quick! 

Are you also working on a text label fix? That’s really what I was trying to get around by putting all my little monitors – all I wanted is a way to label the wires...

Thanks for the amazing work.

Marc

BFH-ktt1, 2021-04-25:
Also the text-bug has already a work-around in head of development, hope this helps.