logisim-evolution/logisim-evolution #557: Copy/paste a text label still fails in 3.4.5

CuriousMarc, 2021-04-25:
Thanks for quickly fixing most of the text label issue. There is still a small part of the bug that's left in 3.4.5 though:
 If you try to copy and paste a text label, it comes up with an error that says:
"Some clipboard components were not pasted because the project libraries do not support them:
  Label"
Admittedly a minor annoyance, now that the text label is mostly working again.
Marc

BFH-ktt1, 2021-04-26:
@CuriousMarc , Thanks for the bug report, I'll look into the details

zdimension, 2021-05-07:
Fixed in fff841, see PR #558

CuriousMarc, 2021-05-07:
> I close this issue as the latest PR of @zdimension fixes the problem, please reopen in case the problem persists

Thanks!

Marc

commit e71da213 (2021-05-07): Merge pull request #558 from zdimension/master

Fix #557 (can't copy labels); add GitHub actions CI
commit fff841d5 (2021-05-07): Fix #557