logisim-evolution/logisim-evolution #561: make aur package for arch linux

Unknown-User2, 2021-05-09:
im on arch and have fun learning how things work with this and would love if you made a aur package for arch linux

maehne, 2021-05-10:
For reference: [Arch Linux Java package guidelines](https://wiki.archlinux.org/title/Java_package_guidelines)

K1ngjulien, 2021-05-15:
It seems like some people have already made AUR packages for this and except for the -git version they look up to date:

https://aur.archlinux.org/packages/logisim-evolution/

https://aur.archlinux.org/packages/logisim-evolution-bin/

https://aur.archlinux.org/packages/logisim-evolution-git/

maehne, 2021-05-19:
Thanks @K1ngjulien for these links!

@Unknown-User2: Could you give them a try and provide us with some feedback?

Based on this, we could add instructions to our [README](https://github.com/reds-heig/logisim-evolution/blob/master/README.md) to point users to these packages and give instructions on how to install them.

Unknown-User2, 2021-05-19:
it work good you just run `logisim-evolution` in the terminal and it runs

maehne, 2021-05-20:
Thanks for the feedback @Unknown-User2. Did you try package `logisim-evolution` or `logisim-evolution-bin`?

MarcinOrlowski, 2021-07-13:
As mentioned, we are unable to officially support Arch for the reasons given [here](https://github.com/logisim-evolution/logisim-evolution/issues/561#issuecomment-836660305). We can add some info to our docs but it should not be outdated - the links given above point to build 3.4.5 with note "This AUR repo has switched to a build-from-source package." whatever that means really. So we need PR from someone that knows Arch. 
Closing as there's not much we can do right now. Feel free to reopen if needed.

closed 2021-07-13 by MarcinOrlowski