logisim-evolution/logisim-evolution #563: VhdlSyntax.java should get auto-generated from VhdlSyntax.flex by gradle

maehne, 2021-05-11:
While reviewing PR #560, I noticed that the repository contains the auto-generated file `src/main/java/com/cburch/logisim/vhdl/syntax/VhdlSyntax.java`. The source file is `src/main/java/com/cburch/logisim/vhdl/syntax/VhdlSyntax.flex`. We have to make sure that we did not commit any changes to `VhdlSyntax.java`, which were not done too to the source file `VhdlSyntax.flex`. Once we are sure that `VhdlSyntax.flex` contains all fixes, we should delete `VhdlSyntax.java` from the Git repository and let `gradle` generate it during the build by running [JFlex](https://jflex.de/). There's a [plug-in](http://jflex-de.github.io/jflex-web/jflex-maven-plugin/) available, which needs to be integrated into our `build.gradle.kts`.

MarcinOrlowski, 2021-06-14:
The PR #673 adds "DO NOT EDIT" warning to generated `VhdlSyntax.java` so at least this issue is somehow flagged.

I found more than one Gradle plugin that is supposed to bridge with JFlex:
* 3 officially [listed on Gradle plugin page](https://plugins.gradle.org/search?term=jflex)
* rusted one (last upd 2016): https://github.com/thomaslee/gradle-jflex-plugin

The JBrains seem IDE bond and patched, so may be not recommended as first choice. The
`org.xbib.gradle.plugin.jflex` one looks more-less up to date (some complained about Gradle 7 compatibility, and [related ticket](https://github.com/jprante/gradle-plugin-jflex/issues/18) is open and not commented) and aside from this people complain about lack of use of Gradle cache and dumping generated files directly into source tree instead of build-dir (possible solution in [the ticket comment](https://github.com/jprante/gradle-plugin-jflex/issues/17)).

Judging from how often flex file changes it may not be worth any further efforts to address this issue for some time and keep the file checked-in and commented.

The only problem I had is that output from Jflex 1.7.0 is invalid (duplicated methods). There are newer versions but the changelog does not contain any entry that could be related. @zdimension @maehne - how do you generate the java class from flex file exactly?

MarcinOrlowski, 2021-07-01:
@zdimension @maehne ping

maehne, 2021-07-01:
I have never generated `VhdlSyntax.java` from `VhdlSyntax.flex` myself. I also never used JFlex myself. I just noticed that it was an auto-generated file. The flex file needs to be updated to incorporate fixes, which actually were done directly to the Java source file.

I agree that `org.xbib.gradle.plugin.jflex` would be probably the first candidate to try to integrate in our Gradle build system for this task. The open bug regarding Gradle 7 might be relatively easy to fix. However, we should first determine how to re-generate the java file manually. Have you tried with latest JFlex version 1.8.2?

davidhutchens, 2021-11-08:
I picked up this issue and started investigating it. I have used the jflex plugin to generate a Java source file. The current source file does not match the generated file. It has been modified by hand. It appears that it was modified originally by replacing a couple of methods, not just some time later.

I have some ideas about how to use jflex to generate Java and incorporate the result into our program without having to edit the file (by hand or automatically). I'll see if I can make it work in a reasonable fashion.

This note is to let everyone know I may be on my way to a solution. If anyone else has worked on it and wishes to discuss it please do.

maehne, 2021-11-08:
Thanks @davidhutchens for working on this issue! Please, note that our Git history is incomplete. Unfortunately, this repo started as a snapshot of a already modified version of the original Logisim. So, some of the modifications to `VhdlSyntax.java` may have landed during this period, in which we don’t have anymore detailed visibility on the made changes.

davidhutchens, 2021-11-13:
I have looked at a number of things. This note is an update on what I have discovered.

The VhdlSyntax files were introduced into LE between 2.15.0 (December 2018) and 3.3.0 (January 2020) from the Holy Cross version. The precise PR doesn't really matter.

They were originally built using TokenTokenMaker, as noted in `VhdlSyntax.flex`. However, I can only find [TokenMakerMaker](https://github.com/bobbylight/TokenMakerMaker), which seems to work in the expected manner. We do not have an input file for TokenMakerMaker.

I can build the `VhdlSyntax.java` file from the `VhdlSyntax.flex` file using jflex 1.4.1 and a skeleton file from TokenMakerMaker. Current versions of jflex do not work. Even the most recent TokenMakerMaker (June 2021) uses jflex 1.4.1.

The only differences in the generated `VhdlSyntax.java` and the one in our project are cosmetic. That is, they are changes to the formatting of the code, but they do not change the functionality. I determined this by carefully examining the diffs and by testing the generated `VhdlSyntax.java` in LE.

I can generate `VhdlSyntax.java` from `VhdlSyntax.flex` in gradle. But there are some things I don't like about my current implementation. I want to think about ways to clean it up before creating a PR.

R3dst0ne, 2021-11-20:
@maehne 
since #1296 got merged, can this issue be closed?

maehne, 2021-11-20:
Yes, this issue is now fixed by PR #1296.

closed 2021-11-20 by maehne