logisim-evolution/logisim-evolution #564: Address issues found in code review of PR #560

maehne, 2021-05-13:
The code review of PR #560 revealed some additional issues, which deserve to get fixed. This issue is created as a reminder, as the PR #560 got already merged due to the many contained improvements. Thanks for them again @zdimension!

zdimension, 2021-05-13:
> Thanks a lot @zdimension for all your work you invested into this PR! Your cleanups are highly appreciated! They make the code more readable and canonical. You also spotted some bugs, which have been lingering in the code base for some time. I had already started reviewing your PR when @BFH-ktt1 merged it. As I found some issues, which deserve a closer look and to be fixed as well, I decided to proceed with the review anyway. It was a bit tedious due to the sheer size of the PR with 468 modified files. Maybe, you could try to submit such refactorings in a bit smaller chunks? I know this can be sometimes hard, as one tends to find more and more issues while performing refactorings.

Apologies about that, I'll try to make smaller PRs in the future, this one is the result of a few days of manually running static analysis inspections and falling into rabbit holes of refactoring, leading the huge number of commits.

> Also a short summary in the PR description can help to steer the code review. Anyway, great work!

I'll add one next time. Glad to be able to help one of my favorite OSS projects!

> I assume you found these issues with the help of some automatic lint tool, e.g., by activating more lint rules using the `-Xlint` options of `javac`. If this is the case, we should add these options to `options.compilerArgs` in `build.gradle.kts` to ensure these kind of issues are not reintroduced unnoticed with future code contributions. Which rules did you activate?

I've used Xlint in the past, but nowadays I mostly rely on IntelliJ inspections and SonarQube "code smells". Although they are often aesthetic, I've found some bugs using them, for example, the xy[1] bug:

    if (xy.length != 2 || xy[0].length() < 1 || xy[0].length() < 1) {

IntelliJ was warning that there was a redundant condition; obviously, the `xy[0].length() < 1` part was present twice. I assumed the two conditions were there on purpose and tried to find a pattern, and notice a similar bit of code a few lines above, which used [0] and then [1].

I added some comments in the PR review, I'll read through everything again in a few days and I'll try to make a PR for the issues you pointed out

maehne, 2021-05-19:
Thanks @zdimension for your feedback. I recognise how much work this refactoring work by yours was. I have done similar efforts in the past and still try when I get to it. Thanks for pointing to IntelliJ and SonarQube for static code analysis. Anyway, we may consider to enable additional Xlint rules for building with gradle to warn more thoroughly about potentially added new issues after code modifications.

davidhutchens, 2026-08-29:
I do not have a strong opinion on this, but I lean toward the view of "if its not broke, don't fix it". So closing this issue works for me.

Do @maehne or @BFH-ktt1 care to comment?

maehne, 2026-09-04:
@hewzhew: I think your proposal to remove the redundant guards is the best way forward to close this issue as resolved.

closed 2026-09-05 by BFH-ktt1