logisim-evolution/logisim-evolution #565: VHDL parser bug: `end entity` results in invalid entity name

datdenkikniet, 2021-05-15:
If I want to import a VHDL entity that looks as follows:

```
entity SomeEntity
    port (input: in std_logic;
            out: out std_logic);
end entity;
```
Logisim tells me that the entity is invalid. I have no clue what the actual VHDL spec says about this, but according to Vivado 2020.1 it is valid.

I'd expect Logisim to be able to import this entity, but it fails and says that the entity name is not valid.

Swapping `end entity` with `end SomeEntity` works as expected.

datdenkikniet, 2021-05-15:
It seems like the issue is that this pattern matcher https://github.com/reds-heig/logisim-evolution/blob/master/src/main/java/com/cburch/logisim/vhdl/base/VhdlParser.java#L166 specifies `END` to _only_ be valid if it is `end SomeEntity`.

maehne, 2021-05-19:
Indeed, the keyword `entity` as well as the entity name are both optional after the keyword `end` in an entity declaration. It is most common to either write just `end;` or `end entity SomeEntity`. Personally, I prefer the latter for entities, architectures, configurations, processes, etc. that may span multiple screen pages.

So, we need to loosen the VHDL syntax checker here.