logisim-evolution/logisim-evolution #566: IntelliJ theme avoids Logisim to launch

Moi78, 2021-05-15:
It is simple : when I set the window theme to intelliJ it just doesn't lauch

Here is the error :
```
Exception in thread "main" java.lang.IllegalAccessError: class com.github.weisj.darklaf.ui.optionpane.DarkOptionPaneUI (in unnamed module @0x1efbd816) cannot access class sun.swing.DefaultLookup (in module java.desktop) because module java.desktop does not export sun.swing to unnamed module @0x1efbd816
        at com.github.weisj.darklaf.ui.optionpane.DarkOptionPaneUI.createButtonArea(DarkOptionPaneUI.java:42)
        at java.desktop/javax.swing.plaf.basic.BasicOptionPaneUI.installComponents(BasicOptionPaneUI.java:205)
        at java.desktop/javax.swing.plaf.basic.BasicOptionPaneUI.installUI(BasicOptionPaneUI.java:159)
        at java.desktop/javax.swing.JComponent.setUI(JComponent.java:685)
        at java.desktop/javax.swing.JOptionPane.setUI(JOptionPane.java:1851)
        at java.desktop/javax.swing.JOptionPane.updateUI(JOptionPane.java:1873)
        at java.desktop/javax.swing.JOptionPane.<init>(JOptionPane.java:1838)
        at java.desktop/javax.swing.ProgressMonitor$ProgressOptionPane.<init>(ProgressMonitor.java:161)
        at java.desktop/javax.swing.ProgressMonitor.setProgress(ProgressMonitor.java:291)
        at com.cburch.logisim.gui.start.Startup.autoUpdate(Startup.java:581)
        at com.cburch.logisim.Main.main(Main.java:74)
```

Here is my java version :
```
java 16.0.1 2021-04-20
Java(TM) SE Runtime Environment (build 16.0.1+9-24)
Java HotSpot(TM) 64-Bit Server VM (build 16.0.1+9-24, mixed mode, sharing)
```

Moi78, 2021-05-15:
(and can someone tell me where is stored the configuration file of the software 🙃 )

Moi78, 2021-05-15:
> (and can someone tell me where is stored the configuration file of the software 🙃 )

(nvm just found it)

maehne, 2021-05-19:
Thanks for reporting this issue @Moi78! On which platform (OS and processor architecture) did you observe this problem?

I suspect it is the same error, which I reported in issue #535. @zdimension: Could you maybe help debugging this issue?

Moi78, 2021-05-19:
Hey @maehne,
The error occured on Windows 10 (x86_64). I think indeed it is the same error as in the issue #535.

I would love to help debugging the issue, but since I'm a C/C++ guy and I only have the basics of Java I'm affraid I can't help that much. Well if there's anything I can do to help, tell me. (I have a JDK & Eclipse installed tho)

zdimension, 2021-05-19:
This seems to be a known issue in Darklaf (https://github.com/weisJ/darklaf/issues/234). Some of their classes are using internal Swing components (explicitely marked as "do not use, API may change"), and Java 16 broke some stuff.

@Moi78 can you try launching Logisim with the `--illegal-access=permit` JVM parameter ? (something like `java --illegal-access=permit -jar logisim.jar`)

Moi78, 2021-05-20:
Well, I tried that, it works ! However, it seems that there's still some bugs... I had some error messages such as : 
`[2021-05-20 10:56:16] [WARNING] Could not load color with id 'Icons.CheckBoxFocused.activeFillColor' fallbacks[]. Using color 'java.awt.Color[r=255,g=0,b=0]' instead. [at com.github.weisj.darklaf.icons.IconColorMapper]`

But at least, it launches.

And I dunno if it is a bug, but it doesn't sound right to me to have wide light elements in a dark mode theme.
[image]

maehne, 2021-05-20:
@Moi78: Could you try to apply the following patch to `build.gradle.kts` and test whether this reliably launches Logisim-evolution with the right JVM options to not trigger the bug?

```diff
diff --git a/build.gradle.kts b/build.gradle.kts
index 29fbfb36..8539c2b1 100644
--- a/build.gradle.kts
+++ b/build.gradle.kts
@@ -17,6 +17,9 @@ repositories {

 application {
     mainClass.set("com.cburch.logisim.Main")
+    // Work around for issue #566 until Darklaf has been updated to not anymore
+    // rely on internal Swing APIs (weisJ/darklaf#234).
+    applicationDefaultJvmArgs = listOf("--illegal-access=permit")
 }

 dependencies {
```

It should be taken into account when launching Logisim-evolution with `gradle runShadow` and also using the executable wrapper installed by a package generated with `gradle jpackage`.

maehne, 2021-05-20:
@Moi78 and @zdimension: My branch [workaround-issue-566](https://github.com/maehne/logisim-evolution/tree/workaround-issue-566) contains the above workaround for testing purposes.

maehne, 2021-06-06:
Thanks @2xsaiko for that helpful suggestion!

2xsaiko, 2021-06-06:
No problem! We replaced the old Darcula library we were using with FlatLaf in a project I work on as well to fix a similar issue I believe, and it works great for us.
Also, the dev is a cool guy and very helpful, he commented on our issues involving FlatLaf or Swing in general a couple times. (Still not sure how he found the project, to be honest, but I'm not complaining!)

Moi78, 2021-06-06:
Hey, I was really busy these last days so I weren't able to test what you asked me to do, and I'm so sorry.

However, I'm glad to see that you found a workaround. Seems to be a good alternive !

zdimension, 2021-06-06:
@Moi78 if you have time, could you try building the commit referenced in #619 ? It uses FlatLaf instead of DarkLaf, which is a bit more stable and should fix the bugs you encountered. (since I wasn't having those bugs before I can't tell for sure if they're fixed, hence why it'd be great if you could test on your computer)

maehne, 2021-06-11:
Closing this issue, as PR #619 got merged. Feel free to reopen if the problem persists.

PR #619 (2021-06-10): Replace DarkLaf with FlatLaf for better compatibility
This probably fixes #566 and #535 (@maehne can you test on your computer, see if the bugs are still there?)
closed 2021-06-11 by maehne