logisim-evolution/logisim-evolution #569: Invalid VHDL netlist generated from circuit when using multiple splitters

maehne, 2021-05-19:
Logisim-evolution does not always correctly netlist circuits to VHDL involving multiple splitters. This can happen in flat and hierarchical circuits.

`flat_top`:

[image]

`hierarcharchical_top`:

[image]

`join_signals`:

[image]

A simple test case with a flat and hierarchical circuit (`flat_top` and `hierarchical_top`, respectively) is attached. I built Logisim-evolution from the current `develop` `HEAD` (commit 32152c1524d81c2d5a62142eee92fa9d27c24533).

[vhdl_netlist_bug.circ.zip](https://github.com/reds-heig/logisim-evolution/files/6509556/vhdl_netlist_bug.circ.zip)

BFH-ktt1, 2021-05-22:
This is related to the VHDL generator of the LED-matrix. The VHDL generator of the LED-Matrix should be completely reviewed.

BFH-ktt1, 2021-08-13:
fixed in my latest PR, closing

commit b998819e (2021-08-14): Fixed dotmatrix HDL generation bug fixes #569
closed 2021-08-13 by BFH-ktt1