logisim-evolution/logisim-evolution #576: Setup SonarQube integration

zdimension, 2021-05-20:
SonarQube is a code quality analysis tool which could be of great use to a project as big as Logisim.

It needs to be hosted on a server to run, but their SonarCloud service is free for open-source projects. Only people with rights on the repository can create the account, maybe @maehne @BFH-ktt1 can look into that? Once created, I can set up the GitHub Actions trigger to launch the analysis.

As an example, I ran an analysis locally; here are the results:

[image]

maehne, 2021-05-20:
I think using continuously such static analysis service will allow us to improve the code quality. I think setting this up will be facilitated by first moving logisim-evolution to an own organisation account as you suggest in issue #577.

R3dst0ne, 2021-11-19:
@zdimension @maehne @BFH-ktt1
Since Logisim-evolution now has an own organization account, is what is the status on this one?

zdimension, 2021-11-19:
I've made a request on the organization, someone with owner perms need to approve it
[image]

BFH-ktt1, 2021-11-24:
@zdimension : What do I exactly need to do, I'm not familiar with sonarqube. Could you send me a link where to approve?

zdimension, 2021-11-24:
> @zdimension : What do I exactly need to do, I'm not familiar with sonarqube. Could you send me a link where to approve?

Here is the link to add an organization and a repository: https://sonarcloud.io/welcome

BFH-ktt1, 2021-11-24:
@zdimension : Just acknowledged the request, it should be setup now, please let me know if I have to do something more.

zdimension, 2021-11-24:
> @zdimension : Just acknowledged the request, it should be setup now, please let me know if I have to do something more.

There are some [additional configuration steps](https://sonarcloud.io/project/configuration?analysisMode=GitHubActions&id=logisim-evolution_logisim-evolution) which require repository admin access, if you grant me management privilèges I can set it up, otherwise a repo admin (either you or Torsten) will have to do it

zdimension, 2021-11-24:
More precisely I need access at least to the "CI Secrets" section to set up the GA integration

zdimension, 2021-11-24:
@BFH-ktt1 Could you grant me the permissions required to manage the repository on sonarcloud, and to the "CI Secrets" section of GH repo settings to set up the GA integration?
[image]

BFH-ktt1, 2021-11-25:
@zdimension : I granted you all the rights. Please let me know when you are done

R3dst0ne, 2023-02-05:
@zdimension @BFH-ktt1 what is the status on this one?

maehne, 2026-09-28:
@zdimension: Could you please also have a look at this issue while looking at issue #1954, as both relate to SonarQube?

zdimension, 2026-09-29:
Yes, will look into it tomorrow

maehne, 2026-09-30:
Thanks @zdimension!