logisim-evolution/logisim-evolution #579: Combinational Analysis: Deleting an input variable causes a ConcurrentModificationException

maehne, 2021-05-20:
If an input variable is selected in the `Combinational Analysis` window and then the key `Delete` is pressed to remove it, a `java.util.ConcurrentModificationException` is thrown:

[image]

The full stack trace is:

```
Exception in thread "AWT-EventQueue-0" java.util.ConcurrentModificationException
        at java.base/java.util.ArrayList$Itr.checkForComodification(ArrayList.java:1013)
        at java.base/java.util.ArrayList$Itr.next(ArrayList.java:967)
        at com.cburch.logisim.analyze.model.TruthTable$MyListener.removeInput(TruthTable.java:666)
        at com.cburch.logisim.analyze.model.TruthTable$MyListener.inputsChanged(TruthTable.java:588)
        at com.cburch.logisim.analyze.model.TruthTable$MyListener.listChanged(TruthTable.java:526)
        at com.cburch.logisim.analyze.model.VariableList.fireEvent(VariableList.java:101)
        at com.cburch.logisim.analyze.model.VariableList.remove(VariableList.java:139)
        at com.cburch.logisim.analyze.gui.VariableTab$2.actionPerformed(VariableTab.java:137)
        at java.desktop/javax.swing.SwingUtilities.notifyAction(SwingUtilities.java:1809)
        at java.desktop/javax.swing.JComponent.processKeyBinding(JComponent.java:2900)
        at java.desktop/javax.swing.JTable.processKeyBinding(JTable.java:5264)
        at java.desktop/javax.swing.JComponent.processKeyBindings(JComponent.java:2948)
        at java.desktop/javax.swing.JComponent.processKeyEvent(JComponent.java:2862)
        at java.desktop/java.awt.Component.processEvent(Component.java:6394)
        at java.desktop/java.awt.Container.processEvent(Container.java:2264)
        at java.desktop/java.awt.Component.dispatchEventImpl(Component.java:4993)
        at java.desktop/java.awt.Container.dispatchEventImpl(Container.java:2322)
        at java.desktop/java.awt.Component.dispatchEvent(Component.java:4825)
        at java.desktop/java.awt.KeyboardFocusManager.redispatchEvent(KeyboardFocusManager.java:1950)
        at java.desktop/java.awt.DefaultKeyboardFocusManager.dispatchKeyEvent(DefaultKeyboardFocusManager.java:875)
        at java.desktop/java.awt.DefaultKeyboardFocusManager.preDispatchKeyEvent(DefaultKeyboardFocusManager.java:1144)
        at java.desktop/java.awt.DefaultKeyboardFocusManager.typeAheadAssertions(DefaultKeyboardFocusManager.java:1014)
        at java.desktop/java.awt.DefaultKeyboardFocusManager.dispatchEvent(DefaultKeyboardFocusManager.java:840)
        at java.desktop/java.awt.Component.dispatchEventImpl(Component.java:4874)
        at java.desktop/java.awt.Container.dispatchEventImpl(Container.java:2322)
        at java.desktop/java.awt.Window.dispatchEventImpl(Window.java:2773)
        at java.desktop/java.awt.Component.dispatchEvent(Component.java:4825)
        at java.desktop/java.awt.EventQueue.dispatchEventImpl(EventQueue.java:772)
        at java.desktop/java.awt.EventQueue$4.run(EventQueue.java:721)
        at java.desktop/java.awt.EventQueue$4.run(EventQueue.java:715)
        at java.base/java.security.AccessController.doPrivileged(AccessController.java:391)
        at java.base/java.security.ProtectionDomain$JavaSecurityAccessImpl.doIntersectionPrivilege(ProtectionDomain.java:85)
        at java.base/java.security.ProtectionDomain$JavaSecurityAccessImpl.doIntersectionPrivilege(ProtectionDomain.java:95)
        at java.desktop/java.awt.EventQueue$5.run(EventQueue.java:745)
        at java.desktop/java.awt.EventQueue$5.run(EventQueue.java:743)
        at java.base/java.security.AccessController.doPrivileged(AccessController.java:391)
        at java.base/java.security.ProtectionDomain$JavaSecurityAccessImpl.doIntersectionPrivilege(ProtectionDomain.java:85)
        at java.desktop/java.awt.EventQueue.dispatchEvent(EventQueue.java:742)
        at java.desktop/java.awt.EventDispatchThread.pumpOneEventForFilters(EventDispatchThread.java:203)
        at java.desktop/java.awt.EventDispatchThread.pumpEventsForFilter(EventDispatchThread.java:124)
        at java.desktop/java.awt.EventDispatchThread.pumpEventsForHierarchy(EventDispatchThread.java:113)
        at java.desktop/java.awt.EventDispatchThread.pumpEvents(EventDispatchThread.java:109)
        at java.desktop/java.awt.EventDispatchThread.pumpEvents(EventDispatchThread.java:101)
        at java.desktop/java.awt.EventDispatchThread.run(EventDispatchThread.java:90)
```

Removing output variables works without triggering any error.

simoc, 2021-05-30:
This looks like the same bug as #511.

I already fixed that bug in branch `develop`, but recent commit ba0080ec962c903a347cfed13746a9078fab61e9 removed my changes and introduced the bug again.

maehne, 2021-05-31:
Thanks @simoc for diagnosing (again) the cause for this bug! PR #607 reapplies your fix.

maehne, 2021-06-11:
Closing since PR #607 got merged.

closed 2021-06-11 by maehne