logisim-evolution/logisim-evolution #580: Replace outdated lablib-checkboxtree-3.3.jar dependency

maehne, 2021-05-20:
`lablib-checkboxtree-3.3.jar` is our last dependency, we distribute in binary form together with the Logisim-evolution sources. It would be nice to just reference the dependency from Gradle. Unfortunately, `lablib-checkboxtree-3.3.jar` is not available on [Maven Central](https://search.maven.org/), but a [3rd-party repository on GitHub](https://github.com/nroduit/mvn-repo/) was found. There's a certain risk that this one will be deleted by its maintainer. We could consider forking it to just maintain this last external and outdated JAR dependency until it can be replaced with a suitable alternative. However, this implies some effort without much merit. The upstream developers of CheckBoxTree did not react to [my request](https://github.com/lorebiga/CheckboxTree/issues/2) to fix the issue on their.

Therefore, I propose to replace this dependency with an alternative instead of relying on the 3rd-party repository for fetching it, as proposed in my earlier PR #533. Actually that dependency is used in a single place:

```
$ grep -rHn checkboxtree src
src/main/java/com/cburch/logisim/gui/appear/ShowStateDialog.java:48:import it.cnr.imaa.essi.lablib.gui.checkboxtree.CheckboxTree;
src/main/java/com/cburch/logisim/gui/appear/ShowStateDialog.java:49:import it.cnr.imaa.essi.lablib.gui.checkboxtree.TreeCheckingModel;
```

[Searching for an alternative](https://www.google.com/search?q=java+swing+checkbox+tree) yields some results, e.g.:

- [this stackoverflow question](https://stackoverflow.com/questions/21847411/java-swing-need-a-good-quality-developed-jtree-with-checkboxes)
- [swing-checkbox-tree](https://github.com/scijava/swing-checkbox-tree) (under BSD-2 license)

I propose to have a closer look to [swing-checkbox-tree](https://github.com/scijava/swing-checkbox-tree) and adapt our code in `ShowStateDialog.java` to it.

davidhutchens, 2021-06-03:
I concur with @maehne. At first glance it does look like swing-checkbox-tree might work. I'll see if I can implement that replacement.

MarcinOrlowski, 2021-06-23:
It looks like this #709 may benefit from the change as well.

MarcinOrlowski, 2021-06-23:
> We could consider forking it to just maintain this last external and outdated JAR dependency until it can be replaced with a suitable alternative

If this lib poses a risk, can't we just add its to LS' sources and build as part of the project? Wouldn't make much difference, with clear benefit of not being nor external nor binary.

davidhutchens, 2021-06-23:
I have pretty much developed a replacement for the lablib-checkboxtree dependency. I hope to put out a PR soon. It would be nice to have the recent proposed changes to src/main/java/com/cburch/logisim/gui/appear/ShowStateDialog.java by @MarcinOrlowski incorporated first so I can deal with any merge issues.

MarcinOrlowski, 2021-06-30:
@maehne can this one be closed as per @davidhutchens merged PR?

maehne, 2021-06-30:
Thanks @davidhutchens for your PR fixing this issue!

closed 2021-06-30 by maehne