logisim-evolution/logisim-evolution #581: Set up a Gitter chatroom (or other platform)

zdimension, 2021-05-20:
I think it could be of use to the project to provide a chatroom related to Logisim, questions and issues with it. Maybe a Gitter space or a Discord server?

BFH-ktt1, 2021-05-22:
I'm in favor. However, I will not have any bandwidth till June/July due to a very busy teaching schedule. After this I will look into moving logisim to it's own namespace and what can be done with respect to a chatroom. Thanks for your suggestion.

maehne, 2021-05-23:
I think [Gitter](https://gitter.im) would be a good option. I am already using it for other projects. With the [developing integration into the Matrix ecosystem](https://blog.gitter.im/2020/09/30/gitter-element-acquisition/), the service is also much more usable on mobile platforms using the Element chat client.

zdimension, 2021-05-26:
Indeed, it would also potentially have a better integration with other GitHub features.

maehne, 2021-05-26:
@zdimension: OK, then let's try to go with the Discussions feature, which @BFH-ktt1 already enabled. If you feel it's not sufficient and a real chat platform would be better, feel free to reopen this issue and we can discuss this topic further.

closed 2021-05-26 by maehne