logisim-evolution/logisim-evolution #582: Simulation frequency very unstable

zdimension, 2021-05-20:
I drew this very simple circuit:

[image]

I then started the simulation and measured the effective frequency compared to the target frequency (in the menu). Here are the results:

[image]

I also noticed that dragging (click+move) the mouse on blank space in the circuit made the effective frequency a lot higher.

As an example, [this video](https://www.youtube.com/watch?v=7Sxhc7uNSA8) shows the circuit running at 40 Hz (target = 256). Simply clicking and moving the mouse makes the running frequency go around 100-110Hz (and the music is played faster). As soon as I stop doing it, it falls back to 40.

At first, I thought the buzzers, or the ROM, or the counter, were causing this, but as you can see, a simple circuit made of a clock and a wire are sufficient to highlight the issue.

BFH-ktt1, 2021-05-22:
I also observed this behavior. It might be related to the fact that I did not merge yet all improvements made by Kevin Walsh (he drastically rewrote the simulation core), it might also be another issue. I'll keep track of it.

maehne, 2021-07-12:
The enhancement proposed in issue #786 would probably help to fix this issue.