logisim-evolution/logisim-evolution #585: Issue with 64 bit roms save/load

zerozen, 2021-05-23:
I have been able to repeat this issue with 64 bit roms, but not other widths. Using 3.4.6 and also tried the latest dev branch (version appears to be the same)

Put a rom with any address size (8 is fine) and a data bit width of 64. Enter "8000000000000000" as a value anywhere. The rom will function as expected, but when you save the content of the ROM to a text file and load it it will not recognize this value as valid.

"7000000000000000" works fine. "0a00000000000000" also works but not "a00000000000000". Also tried all 9's and F's. It's like the leftmost hex character won't go over 7.

Error produced when loading the ROM from file:
There was one error encountered decoding with this format:
Line 2: Data word "8604000000000000" contains non-hex characters.

maehne, 2021-05-23:
Thanks for the report @zerozen. Support for 64 bit wide signals in Logisim-evolution was added only recently and we are aware that some code parts might still assume that signals can be at maximum 32 bits wide. You have stumbled on such a case. The behaviour observed by you smells like the data gets stored in a signed 64 bit integer and that data is tested to be positive.

BFH-ktt1, 2021-05-24:
Is fixed by @zdimension PR. Closing the issue

commit b0f0ac29 (2021-05-24): Merge pull request #587 from zdimension/fix-64-bit-read

Fix #585 (64-bit rom loading)
commit cb7332bc (2021-05-24): Fix #585 (64-bit rom loading)
closed 2021-05-24 by BFH-ktt1