logisim-evolution/logisim-evolution #590: Update the "Credits" list

zdimension, 2021-05-24:
Since it was last updated, there have been lots of contributions from lots of new contributors, I think the list should be updated to reflect that

BFH-ktt1, 2021-05-25:
@zdimension , I fully agree, but for that I need the information that those contributors have to give me. I am very happy to add them to the "credits" list, or that they add themselves and do a PR.

R3dst0ne, 2023-02-05:
Is this issue still relevant?
Since this issue got created, the most active contributes have been added.
In addition, `credits.md` features the line `If you feel that your name should be in this list, please feel free to open a ticket.`.

maehne, 2023-06-06:
@R3dst0ne: I agree that we can close it, as the credits got updated by several PRs in the past months.

closed 2023-06-06 by maehne