logisim-evolution/logisim-evolution #596: Do you consider making an online version? (no install)

kevduc, 2021-05-26:
If so would you use Javascript?

BFH-ktt1, 2021-05-26:
For the moment there are no plans to make an online version (due to limited bandwidth and still a stack of things to be realized). However, it might be a thing to think about in the future.

zdimension, 2021-05-26:
Maybe GWT could be used for that? [Circuit](http://www.falstad.com/circuit/) was ported from Java to the Web that way

maehne, 2021-05-26:
Maybe [CheerpJ](https://leaningtech.com/cheerpj/#Download) is another alternative to consider for making Logisim-evolution available as a Javascript+WebAssembly-based web application.

maehne, 2021-05-26:
Some further options are discussed in [this stackoverflow question](https://stackoverflow.com/questions/440302/how-to-cross-compile-java-source-code-to-javascript). For us it is low priority. PRs going in this direction are welcome as long as Logisim-evolution can still be compiled as a cross-platform stand-alone desktop application.