logisim-evolution/logisim-evolution #601: M+++

tcpipchip, 2021-05-28:
About 8 years ago we developted a processor using the the OLD LOGISIM.
I always use it on my classroom.

I would like to know if someone is interested at free time, to port it to LOGISIM evolution and maybe, port to FPGA, that is my Dream and use in classroom! :)

[image]

Thank you!

Miguel Wisintainer

maehne, 2021-05-31:
I cannot offer help for actively porting your design. However, porting your design to Logisim-evolution and making it synthesisable to FPGA should be doable with reasonable effort. Have you already tried opening your old `.circ` file in Logisim-evolution? Automatically, the old component shapes should get used in most places, so that only few places should require manual fixing. The FPGA Commander will complain if your design contains components, which are not supported for synthesis. These will have to be replaced -- or you or another submits a PR for supporting them. ;-) Once everything works, you may consider changing the look of the part shapes to the more modern default look.

Looking at your top-level circuit, I mainly see one issue: ROM and RAM drive the same data bus, which also seems to be used as input and output for the RAM. This part will require a redesign for making it compatible with synthesis. Personally, I would suggest to merge as much as possible signals to busses, as this facilitates wiring and reduces clutter on the schematic. For visualising multi-bit input/output, you could use the DIP switch and hex digit display.

Which FPGA board/family you would like your design to be ported? Currently, we support the Intel and Xilinx toolchains and quite a long list of popular FPGA boards.

tcpipchip, 2021-06-01:
Please,
@maehne 
How are you ?
Thanks for your comments!

Please, see the final version!

[](http://www.inf.furb.br/~maw/mais/M+++%20LITE.zip)

See if it´s better to implement!

Yes, i replace all the that the Logisim evolution cant convert!

please, send me your comments about!

tcpipchip, 2021-06-03:
Good @maehne  and team

I am converting our M+++ processor to Logisim evolution and going very well. I already can execute all the instructions on EVOLUTION.

But there is a question.

This is the original STACK POINTER

[image]

No i am converting to Evolution

[image]

Am i doing something wrong ? Or some tip, i am not getting the same result during the RAM write on EVOLUTION. On old Logisim works very very nice!

Thank you!

tcpipchip, 2021-06-03:
[Mmaismais2.0.4-Exemplo01.circ.zip](https://github.com/logisim-evolution/logisim-evolution/files/6591790/Mmaismais2.0.4-Exemplo01.circ.zip)
[Mmaismais2.0.4-Exemplo01.circ.zip](https://github.com/logisim-evolution/logisim-evolution/files/6591796/Mmaismais2.0.4-Exemplo01.circ.zip)
One is on OLD and other is on EVOLUTION.

Thank you!

Btw, i plan to use this

https://pt.aliexpress.com/item/1005001678166262.html?spm=a2g0o.productlist.0.0.1652793cRDpOtq&algo_pvid=a582dad5-5b72-444d-b96e-0f89b039a143&algo_expid=a582dad5-5b72-444d-b96e-0f89b039a143-4&btsid=0b0a556a16225643231754896efd3b&ws_ab_test=searchweb0_0,searchweb201602_,searchweb201603_

tcpipchip, 2021-06-03:
Our code to run on ROM

	MOV 00,A;
SOMA:
	INC A,A;
	JMP SOMA;

Binary code to put on ROM

[teste.zip](https://github.com/logisim-evolution/logisim-evolution/files/6591832/teste.zip)

maehne, 2021-06-04:
For your design being compatible with synthesis, you'll have to avoid using tristate signals internally. Remove the tristate buffers from your design. Keep the outputs of the RAMs always enabled. For synchronous design, all FFs, registers, RAMs, and ROMs should get connected to one single clock. You then control if a write should happen either by activating the corresponding WE input or (in the case of FFs), by feeding back either the current output value or the new value depending on some selection signal.

If you have a data bus with multiple drivers, you'll have to replace it with some multiplexer, which selects the correct input value and forwards it.

**PS:** By the way, this issue is a bit off-topic, as it concerns mostly the porting of your particular design and not an issue in Logisim-evolution's implementation. Next time, I suggest you for such cases to open a thread in the [Discussions forum](https://github.com/logisim-evolution/logisim-evolution/discussions) instead. However, there's no need that you close this issue and restart its discussion in the Discussions forum (since it's read by the same people).

tcpipchip, 2021-06-04:
Hi, starting removing all BUFFER TRISTATE, i am using the MUX with OE to replace them!

zdimension, 2021-06-06:
@maehne I don't have the privileges for that on this repo, but if my memory is correct an issue can be converted to a discussion:
[image]

tcpipchip, 2021-06-06:
yes, that memory "problem" that i want to solve now,  that is only the problem that i got during converting to EVOLUTION.
I still wants to present it to classroom the M+++ on evolution!

closed 2021-06-11 by maehne