logisim-evolution/logisim-evolution #611: VHDL components do not preserve signed and unsigned

BFH-ktt1, 2021-06-02:
When adding a VHDL component with inputs/outputs of type signed or unsigned they are transformed by the HDL-writer with std_logic_vector.

BFH-ktt1, 2021-10-05:
I'll close this issue, as the whole base of logisim is based on std_logic and std_logic_vector, introducing signed and unsigned as option will be a major redesign of the HDL-generator that is in my opinion not worth it.

maehne, 2021-10-07:
I understand why the HDL writer relies mostly on `std_logic_vector`. However, I still think that it would be good that we promote/support typesafe entity interfaces for the VHDL component. One way this could be achieved in a relatively simple way would be to generate around a `user_component` and `user_component_wrapper`, which has exactly the same ports as the `user_component`, where the types are just changed to `std_logic_vector` and appropriate casts are done inside the wrapper. To discuss this, I reopen this issue.

BFH-ktt1, 2021-10-17:
I solved it in the moment by throwing an exception that informs the user that only `std_logic` and `std_logic_vector` are accepted, this way the user can do the type-casting by him/herself. Doing something like a wrapper is more trickier, it would work for `unsigned` and `signed`, but what about all the other types around in VHDL. Closing the issue for now.

closed 2021-10-17 by BFH-ktt1