logisim-evolution/logisim-evolution #617: Changing Splitter's fan out param resets custom bit mapping

MarcinOrlowski, 2021-06-04:
Changing Splitter's fan out parameters resets custom bit mapping. Steps to reproduce:

- add splitter to your circuit.
- set fan out to 2 (default)
- set bit width in to 4 (or more)
- map all bits but last one to pin 0 (top)
- map last bit to pin 1 (bottom)
- change fan out to 3
- bit mapping gets wrecked

BFH-ktt1, 2021-06-04:
This is wanted behavior and is not easy to change due to the data structure of the splitter.

MarcinOrlowski, 2021-06-04:
Please correct me as I literally spent 5 minutes looking at the source code but at a first glance it looks like making `configureComponent()` aware of previous state would do the trick? If not, can you please elaborate shortly what would bite next?

maehne, 2021-06-05:
I agree with @BFH-ktt1 that it is wanted behaviour that the bits get distributed equally over the configured number of fan out pins. We need to have a predictable behaviour when the number of bits or number of fan out pins get changed (increased or decreased). If you can come up with a straight-forward alternative behaviour, feel free to propose it here. We are also open to PRs in this direction! It's important to us to reduce the friction when drawing schematics with Logisim-evolution.

zdimension, 2021-06-09:
While making circuits, I often encounter the case where I've spent a few minutes configuring a splitter (the UI for that should be reworked, too), and only at the end notice that I missed a bit somewhere and I need to add an out wire and it resets the configuration to the default one. Although it is indeed useful to equally distribute input pins to output pins, there should definitely be a way to force one behavior or the other.

maehne, 2021-06-11:
I agree with @zdimension and @MarcinOrlowski that it's tedious to (re-)configure the splitters. Ideas and PRs improving it are very welcome!

davidhutchens, 2021-06-11:
I also thought that it was tedious to (re-)configure splitters until I realized I could select a lot of wires on the left by dragging over them, or clicking the first and shift-clicking the last. Then I could apply them all to the same output with one selection. So as long as outputs have contiguous wires, it is quite easy. Even if they are not contiguous you can select them with cmd-click (on the Mac, I don't know the dis-contiguous selection key on other platforms) and that isn't too difficult either.

Since selecting many wires isn't obvious in the interface, is there a way to make it more obvious?

It might be easier if added outputs were left empty and the others were left alone. If reducing outputs, pile all the extras into the last. But I'm not sure.

maehne, 2021-06-11:
Thanks @davidhutchens for your tipp and suggestion. I was not aware that you could change more than one attribute at a time. Maybe a hint could be added, which is shown when hovering with the mouse over the corresponding attributes?

davidhutchens, 2021-06-11:
Now that I think about it, could we add two buttons for reconfiguration. They would use the number of wires on each split determined by what it was attached to and configure the connections in sequence. One one map 0 to 0 and the other would map the last bit to 0 reversing the order of the first. That way, you could wire up the parts and press the appropriate button to do the configuration. That would work for the very common case where the split is sequential.

maehne, 2021-06-11:
Taking into account these common use cases as well is good.

maehne, 2021-06-11:
Another issue, which is not limited to the splitter is that since the increase to 64-bit wide busses, using drop-down lists to choose the bit width and selecting bits is tedious, too.

maehne, 2026-10-01:
Sorry @MobiusRoamer for providing feedback to your offer to work on this issue with such a big delay! It was hard keeping track with all the different issues with the big number of incoming PRs. If you are still interested to work on this issue, we are looking forward to your PR!

MobiusRoamer, 2026-10-02:
> Sorry [@MobiusRoamer](https://github.com/MobiusRoamer) for providing feedback to your offer to work on this issue with such a big delay! It was hard keeping track with all the different issues with [...]

No worries, thx for picking this up, I'll have a go!