logisim-evolution/logisim-evolution #618: Shortcut CTRL-K triggers more than once action

MarcinOrlowski, 2021-06-04:
It seems <kbd>CTRL</kbd>+<kbd>K</kbd> gets assigned to trigger two different actions at the same time. Steps to reproduce:

1. get `Clock` to your circuit
2. click on it and then go to `Properties` tab 
3. Click `Shared Label Font`
4. Cancel the font dialog but do not click anything else
5. Press <kbd>CTRL</kbd>+<kbd>K</kbd>

After step 5 you toggle the `Auto-tick enabled` option and at the same time font dialog will open. For step 4 you can select anything, but font dialog can't be missed. For step 1 you can select anything too of course and in such case once you dismiss font dialog, the "clock source selection" dialog will open.

MarcinOrlowski, 2021-06-05:
Update, it's not just about <kbd>CTRL</kbd>+<kbd>K</kbd>. I noticed that in above scenario I can also have the same issue if I replace step 5 with <kbd>CTRL</kbd>+<kbd>S</kbd>

maehne, 2021-06-05:
Thanks @MarcinOrlowski for reporting this bug. I can reproduce it on macOS with latest Logisim-evolution. It is not only restricted to the clock component, but seems to be linked to the label font dialogue, which is also present in other components, e.g. the Pin component. Besides <kbd>Ctrl</kbd>+<kbd>K</kbd> and <kbd>Ctrl</kbd>+<kbd>S</kbd>, also all other keyboard shortcuts are affected, e.g., <kbd>Ctrl</kbd>+<kbd>P</kbd>. We'll try to look into it, but cannot give a time frame when we'll get to it. Further analysis of the cause and PRs are welcome!

maehne, 2021-06-15:
@BFH-ktt1 hinted in [this comment](https://github.com/logisim-evolution/logisim-evolution/pull/645#issuecomment-860193132) to PR #645 a possible cause for this problem:

> [Each] window has a key-handler. In my earlier implementation for the auto-labler I was not aware yet from this handler and implemented it directly like here as extra handler, causing some keystrokes to mal function (see one of the issues). This is why I did not merge these yet.

PR #1978 (2024-04-15): Fix: multiple keyevent actions removal
Fixes #618 

The table in the "Properties" section would enter edit mode on any key press event if an editable cell in the table was clicked. This behaviour resulted in triggering two events simultaneously, causing unintended behaviour.

The fix implemented in this PR ensures that the table enters edit mode only when triggered by a mouse click or the space key. This change resolves the issue as intended.
closed 2024-04-15 by BFH-ktt1