logisim-evolution/logisim-evolution #625: Add dirty marker also to window title

MarcinOrlowski, 2021-06-07:
I believe it would be useful to have DIRTY_MARKER (*) added also to window title whenever project gets dirty as it would improve "visibility" of the dirty state. Moreover, contrary to in-app project tree, application window title is usually also visible in system taskbar, so one can see the project state without opening the app window.

MarcinOrlowski, 2021-06-10:
That should be relatively easy change, once know how to have it implemented properly (I do not, yet ;). As for PRs - I throw a bunch of such on you recently. Would appreciate any feedback before I run out of spare time. Thanks

maehne, 2021-06-11:
Thanks @MarcinOrlowski for your many contributions over the past days! We very much appreciate them! Sorry for the slow reaction, we are in the final weeks of the spring term and the exam period is approaching. Therefore, our bandwidths is a bit limited at the moment, but @BFH-ktt1 found the time to review and merge them.

MarcinOrlowski, 2021-06-12:
Been teaching on the local uni for 8 years, so no worries, I know the pain :)

maehne, 2021-06-27:
Closing due it got fixed in commit https://github.com/logisim-evolution/logisim-evolution/commit/aeb1a5dab05c74886a7a9ad622838eaa75e9757b. Thanks @MarcinOrlowski!

commit 96c9577e (2021-06-25): Adds "dirty" marker to window title. Closes #625
commit aeb1a5da (2021-06-25): Merge pull request #720 from MarcinOrlowski/dirty-marker-in-window-title

Adds "dirty" marker to window title. Closes #625
commit ed9dd813 (2021-06-25): Adds "dirty" marker to window title. Closes #625
closed 2021-06-27 by maehne