logisim-evolution/logisim-evolution #636: Possible inproper ComponentMapInformationContainer for LED

MarcinOrlowski, 2021-06-10:
The LED constructor, in [line 103](https://github.com/logisim-evolution/logisim-evolution/blob/374a72318ecd0c16b7c077a2260753a1ffc2861c/src/main/java/com/cburch/logisim/std/io/Led.java#L103), sets:

    new ComponentMapInformationContainer( 0, 1, 0 )

however LED port is `INPUT`:

    setPorts(new Port[] {new Port(0, 0, Port.INPUT, 1)});

Shouldn't `ComponentMapInformationContainer` be initialized respectively as:

    new ComponentMapInformationContainer( 1, 0, 0 )

I did the change locally but spotted no visual difference in behavior, so not sure it's intentional or something else?

BFH-ktt1, 2021-06-10:
@MarcinOrlowski : No it is correct, as ComponentMapInformationContainer is used in the FPGA-commander. On toplevel a LED is an output, and for example a button is an input. In simulation of logisim it is exactly the opposite, the pin of the led is an input and the pin on the button is an output. It's a bit confusing, I know :-)
Can I close the issue?

MarcinOrlowski, 2021-06-10:
Is this documented somewhere? Code is barely commented so it's trial and error right now for me, for even such simple things :)

BFH-ktt1, 2021-06-10:
No unfortunately the documentation of the internal structures of logisim is not made as it should have been. The part on the FPGA is made by me, and not documented at all (due to lack of time).

closed 2021-06-10 by MarcinOrlowski