logisim-evolution/logisim-evolution #659: Switch cannot be rotated more than 90 degs

MarcinOrlowski, 2021-06-13:
Switch cannot be rotated more than 90 degs, as subsequent rotations makes no difference due to logic in `paintInstance()`. That unfortunately means that you are currently unable to flip placement of input and output pins by rotating 180 degs from netiher horizontal, nor vertical orientation.

R3dst0ne, 2021-07-01:
This is wrong, the location of the pins is set in `updateports()`.
They can be set in every direction and work correctly.
[image]

MarcinOrlowski, 2021-07-01:
On-button labels are not rotating if i.e. north is 1/0 then south should be 0/1 not 1/0

R3dst0ne, 2021-07-01:
I personally prefer the way it is currently implemented.
Since there is only one option, I don't have to look at the labels to know in which state the switch is.
I find this consistency makes it easier and faster to overview circuits with multiple switches in different directions.
With your proposition you always have to look at the label first and then how the switch is toggled.

MarcinOrlowski, 2021-07-01:
> I personally prefer the way it is currently implemented.

I personally do not. 

>  I don't have to look at the labels to know in which state the switch is.

I made some changes to switch so you can have pints not only on the longer sides, but also on shorter ones. It's then illogical to have output where recessed zero is.

> With your proposition you always have to look at the label first and then how the switch is toggled.

If more would complain it can be made configurable.

MarcinOrlowski, 2021-07-01:
Actually it should be configurable from scratch otherwise configuration from your circuit on the left could not be done. So win-win :)

Vadimatorik, 2026-10-14:
The pin positions already follow all four facings in `updateports()`. The remaining request was a way to rotate the on-body 0/1 marks without losing the current layouts.

That is now the Switch attribute **Rotate labels**, default **No**, so existing circuits keep the current artwork.

- **No** — north and south share one drawing, and east and west share the other, as they do today.
- **Yes** — west and south draw that artwork rotated by 180°, so a north-facing 1/0 becomes 0/1 when the switch faces south. East and north are the reference drawing and do not change when the attribute is toggled. The input, output, component label, and the active output stub stay where they are.

Pull request: https://github.com/logisim-evolution/logisim-evolution/pull/3091

Vadimatorik, 2026-10-15:
AI:

The follow-up on https://github.com/logisim-evolution/logisim-evolution/pull/3091 changes the approach.

The Switch is no longer a one-way buffer from a fixed input to a fixed output. Both ends are contacts. When the switch is closed, a value can pass in either direction. When it is open, the switch does not drive either end, so the two sides stay independent. Two different sources on a closed switch conflict, as they would on one wire.

The Rotate labels option described in the previous comment has been removed. Facing only turns the body, and the 0 and 1 marks show the lever position rather than which end is the source.