logisim-evolution/logisim-evolution #668: SplashScreen looks useless

MarcinOrlowski, 2021-06-13:
I noticed that there's a few issues with the SplashScreen thing. First, it is visible for so short time that it basically looks like visual glitch. Second, it reuses AboutPanel, but contrary to what name of method used to obtain AboutPanel instance, which is `getImagePanel()`, said panel also contains whole credits scroll, which is a quite odd thing for a splash screen. But as the it is gone so fast, nobody managed to noticed this so far :) which gets me to the conclusion that splash screen thingie serves no purpose nowadays and can safely be removed. Am I correct?

maehne, 2021-06-20:
Well, showing a splash screen for a short time is quite standard for GUI apps. We could leave the user the choice, whether to show it at startup or not. Anyway, I would be fine to just show the logo and a generic copyright message without a scrolling list of all developers, as we have for this the Help -> About dialogue.

MarcinOrlowski, 2021-06-22:
The trick is that is now shown for shorter time than Batik needs to render SVG logo :)

BFH-ktt1, 2021-06-30:
Can we close this issue?

MarcinOrlowski, 2021-06-30:
Not yet please. As I said, the main issue right now is that for most of the time splash is visible for shorter period than Baitk needs to render the logo (or this is the visual effect). That simply looks bad like something is broken. And I'd like to avoid having i.e. PNG rendition of the logo just for the splash. Will try to figure out something.  If you can please assign this ticket to me.

MarcinOrlowski, 2021-07-11:
As we dropped svg for now splash started to make bit more sense as at least it does not look broken. Still it is shown for too short. I'm closing this though as it's minior issue an the last thing to worry about.

PR #765 (2021-07-03): Remove svg support
* Replaced SVG in-app logo with PNG
* SVG logo (version with fonts and all-curved) moved to `artwork/`
* Added logo related notes to separated README in `artwork/`
* Removed Batik gradle dependency
* Updated *.md to refer to new location of SVG logo
* Marked SVG files as `binary` type, so they are not diff-ed

Closes #709
Closes #668
closed 2021-07-11 by MarcinOrlowski