logisim-evolution/logisim-evolution #678: Message realting to Logisim Evolution during any flatpak update operation

neildarlow, 2021-06-16:
Whenever I update flatpaks from the command line I see the following output. I have no idea whether this is a packaging issue for Logisim Evolution.

Wrong namespace on document element in '/var/lib/flatpak/exports/share/mime/packages/com.github.reds.LogisimEvolution.xml' (should be http://www.freedesktop.org/standards/shared-mime-info)

OS: fedora SilverBlue 34
Flatpak: flatpak-1.10.2-3.fc34.x86_64

MarcinOrlowski, 2021-06-16:
Looks like wrong icon reference. Comes from [LogisimEvolution.desktop](https://github.com/logisim-evolution/logisim-evolution/blob/374a72318ecd0c16b7c077a2260753a1ffc2861c/support/Flatpak/com.github.reds.LogisimEvolution.desktop#L6)

MarcinOrlowski, 2021-06-30:
@neildarlow do you still have this issue? I just installed LS via flatpak and see no problems. If you still face this message, can you provide more details so I can try to reproduce?

@BFH-ktt1 do you maintain Flatpak page of LS?

MarcinOrlowski, 2021-06-30:
Because I see Flathub entry is sourced from this repo: https://github.com/flathub/com.github.reds.LogisimEvolution - is it related or 3rd party (members' nicks tell me nothing)?

maehne, 2021-06-30:
The flatpak was created by @TheEvilSkeleton and discussed in detail in issue #448.

MarcinOrlowski, 2021-06-30:
@TheEvilSkeleton could you please update the Flathub entry's description to match this landing page: https://github.com/logisim-evolution/logisim-evolution/tree/develop as (at least form me) what sticks out in current one is "and recently the development has been officially stopped." which is bad. Also if you could update the screenshot to one of https://github.com/logisim-evolution/logisim-evolution/tree/develop/docs/img it would be great.

Also can you please correct this entry:

> Developer
> REDS institute

@maehne am I correct that for now `support/Flatpak` can be removed?

@BFH-ktt1 shouldn't we have snap/flatpak packages under our control?

maehne, 2021-06-30:
> @maehne am I correct that for now `support/Flatpak` can be removed?

No, see [this message](https://github.com/logisim-evolution/logisim-evolution/issues/448#issuecomment-749684001) by @TheEvilSkeleton.

> @BFH-ktt1 shouldn't we have snap/flatpak packages under our control?

Ideally yes and I think @BFH-ktt1 has at least maintain rights for the associated repository. However, our knowledge and bandwidth is limited for maintaining all the different kind of packages. Also, manually building them on the different platforms for a release also costs time. We should probably strive for creating these packages as much as possible automatically using, e.g., GitHub Actions. As with many things, we are lacking time and knowledge to work this aspect. So help is welcome.

MarcinOrlowski, 2021-06-30:
Ok, I created #743 for that. 

I am also going to close this ticket as a) it works for me now, b) if it does not for OP then it should currently be reported here: https://github.com/flathub/com.github.reds.LogisimEvolution

closed 2021-06-30 by MarcinOrlowski