logisim-evolution/logisim-evolution #680: Some PR vanished

MarcinOrlowski, 2021-06-16:
It seems that changes incorporated by already merged PRs were somehow reversed by subsequent merges. For example, changes from #648 are no longer in develop. Same were for #655 (restored in #677). I am now reviewing the recent merges.

MarcinOrlowski, 2021-06-16:
@BFH-ktt1 can you please stop merging any PRs for a while until we got a picture what going on and what is now merged and what is no more and needs reapply?

MarcinOrlowski, 2021-06-16:
For now it looks like most of the PRs needs reapply. Changes by these seems to no longer be in develop:

- #645 Hitting ESC closes Preferences window now
- #646 Hitting ESC closes most of the frames
- #648 Cleanup of component libraries (re-patched by #681, still in review)
- #651 General code cleanup
- #654 Small library labels cleanup
- #655 Fix problem reading hex values from saved .circ files with a 1 in bit 63. (reapplied by #677 already)

Other seem (at least at first glance) there.

MarcinOrlowski, 2021-06-16:
@BFH-ktt1 #682 reapplies #654. Ready to merge.

MarcinOrlowski, 2021-06-16:
@BFH-ktt1 #683 reapplies (more/less) #651. Ready to merge

MarcinOrlowski, 2021-06-16:
I am not restoring #645 and #646 (were adding support for ESC closing most of the frames) due to doubts the change is safe and not causing any conflicts with other key handlers.

MarcinOrlowski, 2021-06-16:
@BFH-ktt1 I am not going to restore #645 and #646 for now as you raised the concern about possible conflict with other key handling code. Yet this feature is pretty handy to have, so can you please suggest better approach?

BFH-ktt1, 2021-06-16:
@MarcinOrlowski , I completely agree with you that it is very handy to have. Currently I am in the "end-of-the-semester-stress", but as soon as I have some time I'll look into the details of the "keyhandler". Maybe it must not be in there, but it is better to check

BFH-ktt1, 2021-06-16:
@MarcinOrlowski : B.t.w. I also observed that your changes to the splash and the web-site entry have "disappeared". I really liked this additions :-)

MarcinOrlowski, 2021-06-16:
@BFH-ktt1 Will check and restore these, no problem. You may also want to take a look at #667 thread in "Discussions"

MarcinOrlowski, 2021-06-16:
@BFH-ktt1 Help menu/About changes restored in #687

MarcinOrlowski, 2021-06-21:
@BFH-ktt1 could you please review pending PRs, please? The pile blocks further works.

BFH-ktt1, 2021-06-22:
All merged, can we close this issue?

closed 2021-06-22 by MarcinOrlowski