logisim-evolution/logisim-evolution #696: Polish Translation missing

mbaillif, 2021-06-20:
The Polish translation was lost between version 4.1 and 4.2

I recovered it and adapted it to version 3.5.
But just is it a mistake or is it intended before sending it.

mbaillif, 2021-06-20:
Version 3.4.1 3.4.2  :-)

BFH-ktt1, 2021-06-20:
It was definitely not intended. I think that it "got lost" when transforming to the "new form". So thanks a lot for making a PR to get it in again.

MarcinOrlowski, 2021-06-21:
I can proofread this if needed.

mbaillif, 2021-06-26:
The Polish text is back in my latest pull_request.
I still need to add PL in the settings.properties file

There are still some lines to translate for 100%.  
1 in analysis
1 in circuit
5 in fpga
76 in gui
12 in soc
28 in std
2 i tools
1 in dans util
if someone is able to do that it would be great.

MarcinOrlowski, 2021-06-26:
> I still need to add PL in the settings.properties file

I removed it in [this PR](https://github.com/logisim-evolution/logisim-evolution/pull/634) so just add it back. 

> if someone is able to do that it would be great.

I can do that, but I think we can have it updated in separate PR anyway

MarcinOrlowski, 2021-06-30:
Closing as fixed by #725

closed 2021-06-30 by MarcinOrlowski