logisim-evolution/logisim-evolution #698: Code style

MarcinOrlowski, 2021-06-21:
However docs say code shall be formatted with `google-java-format` it is not often the case. Also formatting do not automatically enforce unified coding style. To remedy this I added Checkstyle checker to Gradle in #697 with the idea to have it run as GitHub Action soon. 

In fact this ticket should go in "Discussions" but no-one participates there, so here we go. I selected `google-java` style "just because" (so please object if you think other style is better for any explainable reasons) but nonetheless, once I run it I was literally flooded by complains. So I think it is important to agree on once code style, then reformat the code to match and make GitHub Action guarding that. 

Please comment.

MarcinOrlowski, 2021-06-23:
I think further code styles will be fixed in two steps: first PR just fix formatting - no code change at all and then separate PRs with code changes, method renames etc. That should simplify the reviews. @BFH-ktt1 what you think?

MarcinOrlowski, 2021-06-23:
#710 was made with CheckStyles tuned using `suppressions.xml`:

```xml
<?xml version="1.0"?>
<!DOCTYPE suppressions PUBLIC
        "-//Checkstyle//DTD SuppressionFilter Configuration 1.2//EN"
        "https://checkstyle.org/dtds/suppressions_1_2.dtd">

<suppressions>
    <suppress checks="MissingJavadocMethod"/>
    <suppress checks="NeedBraces"/>
    <suppress checks="JavadocParagraph"/>

    <suppress checks="LocalVariableName"/>
    <suppress checks="ParameterName"/>
    <suppress checks="MethodName"/>
    <suppress checks="MemberName"/>
    <suppress checks="TypeName"/>
    <suppress checks="VariableDeclarationUsageDistance"/>
    <suppress checks="AbbreviationAsWordInName"/>
    <suppress checks="AvoidEscapedUnicodeCharacters"/>
    <suppress checks="EmptyCatchBlock"/>
    <suppress checks="MissingSwitchDefault"/>
    <suppress checks="MultipleVariableDeclarations"/>
    <suppress checks="EmptyBlock"/>
    <suppress checks="EmptyLineSeparator"/>
    <suppress checks="NonEmptyAtclauseDescription"/>
    <suppress checks="SummaryJavadoc"/>
    <suppress checks="OverloadMethodsDeclarationOrder"/>
    <suppress checks="CustomImportOrder"/>
    <suppress checks="LineLength"/>
</suppressions>
```

MarcinOrlowski, 2021-06-23:
@BFH-ktt1 while I was working on `soc/nios2*` I saw a lot of code formatted like this

[image]

For example [Nios2ProgramControlInstructions.java](https://github.com/logisim-evolution/logisim-evolution/blob/b698a76154c8f15e377056ee9237da160c9822fc/src/main/java/com/cburch/logisim/soc/nios2/Nios2ProgramControlInstructions.java#L113). Git blame claims this is your work so my question is if retaining
this formatting is really important. The problem is that while I got the point why you did that, it's very fragile
and makes maintaining, refactoring of related code problematic, so unless there's really any benefit from
having this formatted that way I am unable to see ATM, I think applying regular formatting to this code
would be a better choice in longer run. And if this needs better visibility, maybe block comments next to each
`case` would do the job sufficiently?

BFH-ktt1, 2021-06-24:
@MarcinOrlowski : The code was structured this way for more readability when I wrote it (sorry I am more a C-programmer as a Java guru :-)). For me there is no reason why it could not be restructured.

MarcinOrlowski, 2021-06-24:
As I wrote - I fully understand why would you do that, but this formatting was so fragile, that its better to fix in controllable manner instead of waiting for PR that would mess it even more. #712 does so.

MarcinOrlowski, 2021-07-11:
As we already have style checker plugged onto PR checks no need to keep this issue open as every PR shall gradually improve tge style anyway.

closed 2021-07-11 by MarcinOrlowski