logisim-evolution/logisim-evolution #709: Build Errors in Eclipse on develop branch

Emil-DV, 2021-06-23:
Tonight I refreshed my local copy of the devlop branch, did gradlew run which succeeded but when I went to build using Eclipse (imported Gradle project) I got lots of errors (see below). The master branch builds w/o error using the same process, as did the develop branch a few days ago.

[image]

MarcinOrlowski, 2021-06-23:
Just checked and it indeed happens here too. It looks like Eclipse do not see our dependencies (i. E. XML libs) and then snowball effect happens

MarcinOrlowski, 2021-06-23:
It looks like this is the same type of problem: https://stackoverflow.com/questions/55571046/eclipse-is-confused-by-imports-accessible-from-more-than-one-module

MarcinOrlowski, 2021-06-30:
Closing as this seems to be gone once we dropped said `*.jar` binary. I just successfully built was just able to built freshly checked-out `develop`

Emil-DV, 2021-07-01:
Sadly I am not having the same luck.  I still get the same batch of errors even though I've completely deleted my clone and started fresh.

MarcinOrlowski, 2021-07-01:
What about Eclipse? It's cache maybe or some settings? I used freshly downloaded this exact version https://www.eclipse.org/downloads/packages/release/2021-06/r/eclipse-ide-java-developers (on Linux).

Emil-DV, 2021-07-01:
Other than being on windows, I am using the same release:
Eclipse IDE for Java Developers (includes Incubating components)
Version: 2021-06 (4.20.0)
Build id: 20210612-2011

I have a linux laptop that I can try it on later today.

MarcinOrlowski, 2021-07-01:
I mostly mean - is this your **clean** Eclipse setup? Can you install i.e. on separate user account? Because I believe OS is rather irrelevant really here.

Emil-DV, 2021-07-02:
I tried it out on another windows machine that has never had Eclipse/any java dev on it and got the same result. The master branch builds okay but the dev branch gives me those same errors.  I'm not married to Eclipse, if there is a better build/debug solution I'm open to trying that.

davidhutchens, 2021-07-03:
I have reproduced this problem on the Mac.

From my research, it seems that the org.apache.xmlgraphics:batik-swing:1.14 library defines (or depends on other libraries that define) things in org.w3c.dom which is reserved for the java libraries. When Eclipse loads the project, it puts the java libraries in the java.xml module in the ModulePath and the projects libraries in the <unnamed> module in the ClassPath. Hence the conflict. It uses that when doing its own build and debug.

Curiously, even as Eclipse is complaining about the files, it is happy to run the gradle build tasks and thus build and run the project. But Eclipse's build does not work. At this point, I think I understand why Eclipse is complaining, but I'm less sure why gradle is not.

I'm not sure how to approach fixing this. But I wanted to post what I have observed in case others are looking at it and might be helped.

MarcinOrlowski, 2021-07-03:
Batik was a step forward to have SVG icons and other in-app graphics but for now it only handles the logo :) so I think in current state it can be safely removed, which I will now.

MarcinOrlowski, 2021-07-03:
@davidhutchens can you please check if #765 resolves this problem?

davidhutchens, 2021-07-03:
I confirm that #765 works in Eclipse on my Mac.

I also note that Eclipse isn't easy to use with respect to dependency changes in build.gradle.kts. It does not update its classpath when the file is changed. I had to remove and recreate the Eclipse project, including removing Eclipse files from both the workspace (with delete in Eclipse) and the project files (bin, .classpath, .project) with rm. I'm not sure how much of that removal was actually necessary as I just cleaned up everything. I noticed this yesterday when just checking out master and refreshing Eclipse did not fix the problem.

After sleeping on it, I think the problem is that Eclipse is not being careful about how it includes dependencies from gradle and is doing something incorrectly that causes this issue. That would explain why gradle itself continues to work, even from inside Eclipse.

MarcinOrlowski, 2021-07-03:
Closing, as #765 is now merged.

closed 2021-07-03 by MarcinOrlowski

Emil-DV, 2021-07-03:
Thanks for this, I'm far more familiar with Eclipse (vs. NetBeans/IntelliJ) from my work with embedded systems (VxWorks, DEOS, etc) so it is good that I am able to use it here as well.  I confirmed that it is working on the Windows version as well.

PR #765 (2021-07-03): Remove svg support
* Replaced SVG in-app logo with PNG
* SVG logo (version with fonts and all-curved) moved to `artwork/`
* Added logo related notes to separated README in `artwork/`
* Removed Batik gradle dependency
* Updated *.md to refer to new location of SVG logo
* Marked SVG files as `binary` type, so they are not diff-ed

Closes #709
Closes #668