logisim-evolution/logisim-evolution #721: Project name - can it be just "Logisim Evolution"?

MarcinOrlowski, 2021-06-24:
Since Logisim-evolution seems to stick as project name for long time and apparently noone got better idea, I suggest to make it official (unless changed :) and normalize spelling etc. My proposal is to use `Logisim Evolution`. Without any hyphen, spelling as shown. It sounds the same but looks better :) I am in process of clearing documentation right now and would like to make project name unified, so please speak-up. @BFH-ktt1?

davidhutchens, 2021-06-24:
The project name is used in various ways. It is in the name of the shadowjar and the distribution files. I'm not sure what all the issues would be with putting a space in the middle of it. Also, some expressed a desire to leave the shadowjar with a lowercase first letter. At least that was how I interpreted the comment. That change would, in theory, also change the URL for the project itself.

I would not personally be opposed to trying to use that form everywhere. I just note that there are a lot of things to test. It might not be so simple.

BFH-ktt1, 2021-06-24:
Some things you should change, some your should not. Since 2012 we stick with logisim-evolution, and it is something which is "burned into the mind". I'm for changes, but this one I would prefer not to change.

MarcinOrlowski, 2021-06-24:
I do not mind either way (also @davidhutchens have a valid point about the rists). So can we agree about the proper spelling at least, as you say `logisim-evolution`, docs read `Logisim-evolution`. So what's the correct one? `logisim-evolution`? `Logisim-evolution`? `logisim-Evolution`? :)

davidhutchens, 2021-06-25:
In #523, @maehne updated the README and noted: 'This pull requests improves wording and formatting of the README. It also proposes to use the casing "Logisim-evolution" instead of "logisim-evolution", as the former is used in the application itself and the help files.'

So that would suggest that Logisim-evolution is the form used in the documentation and help files. I suggest sticking with that in those places and leave the official project name (and build system) with the lower-case first letter.

MarcinOrlowski, 2021-06-25:
The only thing I dislike with the hyphen form is this:

[image]

closed 2021-06-25 by MarcinOrlowski