logisim-evolution/logisim-evolution #731: EEPROM Memory

Emil-DV, 2021-06-27:
The micro-controller I am building could benefit from some form of "EEPROM" memory. A "ROM" with an input port and WE.
Currently I am using a RAM that is marked non-volitile which doesn't clear its contents upon Cntl+R Reset but does between file save/re-open.
Perhaps this has already been done and is in some importable library package of components?
If not I thought that I could either:
0) Create a new "PROM" component by adding the input port and WE functionality to a copy of the current ROM component. (not to include the complex erase/program logic that is in real EEPROMs/Flash memorys)
1) Modify the stock ROM object to have "Programmable" option that adds an input port and WE pin/logic. 
2) Modify the stock RAM object such that those marked "non-volitile" save their contents to the .circ file thus making it a sort of "Battery Backed" RAM.

MarcinOrlowski, 2021-07-13:
@maehne can we have "Evaluating" or "Triage" label created to mark tickets like this one?

maehne, 2026-07-14:
The discussion in PR #2700 showed that it is difficult to implement the behaviour described above in a way, which wouldn't surprise users, and that any implementation of safe guards would require considerable effort without a clear perspective to achieve a satisfying user-friendly and robust result. I am therefore closing this issue as won't fix.