logisim-evolution/logisim-evolution #735: Stale branches `revert-217-dev-with-fsm` and `version`

MarcinOrlowski, 2021-06-30:
Is there any purpose `revert-217-dev-with-fsm` and `version` branches still do exist as of today? These are far behind, so most likely can be safely deleted.

BFH-ktt1, 2021-06-30:
The revert-217-dev-with-fsm is important as it adds fsm-drawing support to logisim. We did not have the time to finish this one.
Version I do not even know what it is.

MarcinOrlowski, 2021-06-30:
Version I do not even know what it is.> 

@mbaillif could you please cast any light on what is/was the `version` branch for?

MarcinOrlowski, 2021-06-30:
> The revert-217-dev-with-fsm is important as it adds fsm-drawing support to logisim. We did not have the time to finish this one.

It's 11 mo old and 450+ commits behind so quite huge tech-debt as of today. I also got a feeling that with current development and refactoring it might face quite massive conflicts. Do you think you could i.e. allocate say 30 mins and see it the works on that branch could still be resumed? Also is there any doc/ticket/whatever that describes what's there to be done, how much left etc?

BFH-ktt1, 2021-06-30:
In the moment, unfortunately, I have very limited bandwidth. It'll be better in about three weeks, I'll put it on my todo list.

mbaillif, 2021-06-30:
On my side I only edit the documentation, and I never worked on two branches.
At some point one branch was modified and I switched to develop. 

Unfortunately I don't have the corresponding email anymore.

There was a big restructuring of the code merged to #203.
And when I look at the code (23.12.2018) that is not reassembled in develop, it seems obsolete.
I think it is only the ashes of this move.

For me, I will delete after checking my statement about #203

maehne, 2021-06-30:
Regarding the goal to add an FSM editor, you may have a look at issue #6. @sderrien added a nice editor in his fork, but we were not yet able to merge it back into our repository, because both forks have significantly diverted. Especially, regarding the build system (Ant vs. Gradle) and many new features, which were back-ported by @BFH-ktt1 from @kevinawalsh's Holy Cross fork. Another challenge were the 3rd-party dependencies Xtext and EMF, with which we are not familiar. 

@MarcinOrlowski: Would you be interested in back-porting this feature?

MarcinOrlowski, 2021-06-30:
Unfortunately I will not have the enough time for that. BUT, maybe @sderrien could drive this effort? He knows this thing best and if any additional resources will be needed then we can assist at specified tasks? @sderrien how you find that?

MarcinOrlowski, 2021-07-13:
Given the above I believe we can safely delete `version` branch.

maehne, 2021-07-13:
I agree, as `version` does not contain any commits, which aren't already present in `develop` and `master`. @BFH-ktt1: Do you agree as well?

R3dst0ne, 2023-02-05:
Since the two branches this issue is about do not exist anymore, can it be closed?

maehne, 2023-03-07:
> Since the two branches this issue is about do not exist anymore, can it be closed?

As these branches are not present anymore and there was no reaction, I am closing this as completed.

closed 2023-03-07 by maehne