logisim-evolution/logisim-evolution #742: HDL support for divider is unfinished

maehne, 2021-06-30:
Although the HDL-generator class is in the working tree, it has never been tested on correct functionality. This is the reason it is not enabled. It is on the todo list and will be enabled as soon as correct functionality is guaranteed.

_Originally posted by @BFH-ktt1 in https://github.com/logisim-evolution/logisim-evolution/issues/430#issuecomment-670605221_

MarcinOrlowski, 2021-07-13:
How much work would that require?

maehne, 2021-07-13:
Without looking at the actual code, a rough guess would be something like half a day to one day. The divider is purely combinatoric, but generic and supports unsigned and signed input values. So, we should ideally create some self-checking test bench for the generated VHDL code.

maehne, 2026-10-12:
Thanks @hewzhew for looking into this issue! I am fine with the suggested approach. PR #3057 by @zOnlyKroks addresses a related topic. @BFH-ktt1: What is your opinion on this?

Regarding a self-checking VHDL/Verilog test bench for this (and potentially the other arithmetic blocks): I think this is a larger separate project, which would require quite some effort to invest.

Currently we lack any automatised framework for testing the generated HDL code. A first step could be creating test circuits, which instantiate the arithmetic components in a minimal test harness (e.g., DIP switches to set the input values the output results passed through a bin-to-BCD and 7 segment decoders to display it with any additional status outputs attached to LEDs), which would allow manual / semi-automatic testing on an FPGA board. This would also test the logic synthesis of the HDL description. The .circ file could contain several test benches instantiating the arithmetic block with different wides that could be synthesised/tested independently.

A more involved test harness could automatically apply different input combinations and check for the results against the synthesisers default implementation for the arithmetic operators and count observed deviations. 

For more thorough checking of the generated HDL code, self-checking test benches could be created - ideally profiting of some open verification framework like OSVVM, Vunit, etc. to check for functional equivalency between the synthesizable architecture and a purely behavioural description of the Logisim component.

zOnlyKroks, 2026-10-12:
> 5\. Thanks [@hewzhew](https://github.com/hewzhew) for looking into this issue! I am fine with the suggested approach. PR [#3057](https://github.com/logisim-evolution/logisim-evolution/pull/3057) by [@zOnlyKroks](https://github.com/zOnlyKroks) addresses a related topic. [@BFH-ktt1](https://github.com/BFH-ktt1): What is your opinion on this?
>     Regarding a self-checking VHDL/Verilog test bench for this (and potentially the other arithmetic blocks): I think this is a larger separate project, which would require quite some effort to invest.
>     Currently we lack any automatised framework for testing the generated HDL code. A first step could be creating test circuits, which instantiate the arithmetic components in a minimal test harness (e.g., DIP switches to set the input values the output results passed through a bin-to-BCD and 7 segment decoders to display it with any additional status outputs attached to LEDs), which would allow manual / semi-automatic testing on an FPGA board. This would also test the logic synthesis of the HDL description. The .circ file could contain several test benches instantiating the arithmetic block with different wides that could be synthesised/tested independently.
>     A more involved test harness could automatically apply different input combinations and check for the results against the synthesisers default implementation for the arithmetic operators and count observed deviations.
>     For more thorough checking of the generated HDL code, self-checking test benches could be created - ideally profiting of some open verification framework like OSVVM, Vunit, etc. to check for functional equivalency between the synthesizable architecture and a purely behavioural description of the Logisim component.

If you are interested, I could draw up a rough idea of how I'd design and implement such a framework and later, if said idea is deemed good enough, also implement it. I agree that this would be a major and larger change

maehne, 2026-10-12:
@zOnlyKroks: Thanks for volunteering to work out a plan to implement a verification framework for the HDL code generated from the Logisim circuit components. Please create a dedicated issue to propose/discuss/track implementation of this framework.

@hewzhew: We also appreciate any input from your side on this issue.