logisim-evolution/logisim-evolution #743: Have own Flatpak/Snap packages

MarcinOrlowski, 2021-06-30:
The task is to have own, fully controlled Flatpak/Snap packages. There's currently 3rd party driven, Flatpak package: https://www.flathub.org/apps/details/com.github.reds.LogisimEvolution driven by @TheEvilSkeleton however at the end of the day we should have full control of our packages as this is the only way to keep it up to date with official releases (up to now we rely on 3rd party to have a time to package new version once it is released).

This ticket is kind of followup to #678

TheEvilSkeleton, 2021-06-30:
@MarcinOrlowski that's perfectly fine. If you want to maintain the Flatpak package over at Flathub, you can email flathub@lists.freedesktop.org. I highly recommend every member/maintainer of this repo to email the Flathub maintainers so they will give you write permission to the repo. ([Source](https://github.com/flathub/flathub/wiki/App-Submission#someone-else-has-put-my-app-on-flathubwhat-do-i-do))

TheEvilSkeleton, 2021-06-30:
I'm going to submit a merge request to the `develop` branch for preparation of official support.

MarcinOrlowski, 2021-06-30:
No, no, I do not want to maintain it :) I would be fantastic if you could still keep eye on this and maintain Logisim-evolution Flatpak package as you did so far. You got all the necessary knowledge how to do that the best way. What I'd like to have though is to keep all the Logisim-evolution stuff under one umbrella which means that we'd have i.e. separate repo  say `logisim-evolution-flatpak` (under `logisim-evelution` organization - @BFH-ktt1 can definitely set it up as needed) which would drive the Flathub entry. Same for snap and others (i.e. PPA). This is important because it lets us maintain the continuity of the service if case you i.e. decide you do not care any more.

MarcinOrlowski, 2021-06-30:
@TheEvilSkeleton do you think it's better to have separate repo for Flatpak, or it should be part of main one?

TheEvilSkeleton, 2021-06-30:
@MarcinOrlowski my suggestion is to put as much as possible in the main repo. Flathub wants to be as vanilla as possible, unless some patches can be used to better integrate the application in a Flatpak container, like [some](https://github.com/flathub/com.github.reds.LogisimEvolution/blob/master/com.github.reds.LogisimEvolution.png) [images](https://github.com/flathub/com.github.reds.LogisimEvolution/blob/master/com.github.reds.LogisimEvolution.svg).

TheEvilSkeleton, 2021-07-08:
@MarcinOrlowski can you take a look at https://github.com/flathub/com.github.reds.LogisimEvolution/pull/13#issuecomment-873425538?

MarcinOrlowski, 2021-07-08:
@TheEvilSkeleton Done.

R3dst0ne, 2022-08-23:
There's currently a 3rd party driven, Snap package:
https://snapcraft.io/logisim-evolution-snapcraft
[This](https://github.com/anrouxel/logisim-evolution-snapcraft) repo by @anrouxel belongs to it.

I agree with @MarcinOrlowski that 
> [...] at the end of the day we should have full control of our packages as this is the only way to keep it up to date with official releases (up to now we rely on 3rd party to have a time to package new version once it is released).

TheEvilSkeleton, 2022-08-24:
Just a reminder I'm fully okay with co-maintaining or stepping down as maintainer from the logisim-evolution flatpak.

R3dst0ne, 2022-08-28:
Thanks to @kjcolley7 we now know that LSe is featured in [Homebrew](https://formulae.brew.sh/cask/logisim-evolution).
I created [this](https://github.com/Homebrew/homebrew-cask/pull/130565) PR that updates the URL and changes the name to the correct one.

kjcolley7, 2022-08-28:
Well, brew _IS_ mentioned in this repo's README as one of the supported installation methods for macOS :)

R3dst0ne, 2022-08-28:
> Well, brew _IS_ mentioned in this repo's README as one of the supported installation methods for macOS :)

I totally missed that one 😅

Until now, I thought we only reference the installers found under [releases](https://github.com/logisim-evolution/logisim-evolution/releases).
If we feature Homebrew, what about mentioning Flatpak and Snap to?

R3dst0ne, 2022-08-28:
Just found:
[Chocolatey](https://community.chocolatey.org/packages/logisim-evolution)
[MacPorts](https://ports.macports.org/port/logisim-evolution/details/)
[winget](https://wingetgui.com/apps?id=Logisim-evolution.Logisim-evolution)
and this sketchy looking one
[OnWorks](https://www.onworks.net/de/software/windows/app-logisim-evolution)

MarcinOrlowski, 2022-09-02:
As a note, I built Snap package of LSe. Still need to polish the scripts and plug them into gradle + add GH builder but that's should soon be ready.

anrouxel, 2022-09-02:
Hello,
I had created this project of publication of logisim evolution on Snapcraft, because last year during my studies in BUT Informatique (France) we used Logisim Evolution in progress. I was an Ubuntu user and Logisim Evolution was not available in Snapcraft that's why I published Logisim Evolution on Snapcraft.

Best regards

anrouxel, 2022-09-02:
@MarcinOrlowski I saw that you had forked the deposit that I created. I updated the build script (Snapcraft) from core20 to core22.

MarcinOrlowski, 2022-09-03:
@anrouxel I already have base22 supported, thanks.

MarcinOrlowski, 2022-09-20:
@anrouxel I already established "official" snap for LSe [here](https://snapcraft.io/logisim-evolution) (BTW: thanks for not taking `logisim-evolution` name), but we also need to have snap for our nightly builds. I did some early work to get there with our `.github/workflows` [here](https://github.com/MarcinOrlowski/logisim-evolution/tree/snap-packages-test-1/) but it is broken right now and I am currently out of spare cycles to finish the works. Let me know if you have some spare cycles to help with it. Please let me know. Thanks

R3dst0ne, 2022-10-05:
I noticed two things with the official [Snap Package](https://snapcraft.io/logisim-evolution).
- The current release is still 3.7.2. (edge: 3.8.0-dev)
- The wrong name gets used `Logisim-Evolution` vs the official `Logisim-evolution`.

MarcinOrlowski, 2022-10-05:
I know. Will update both shortly.

maehne, 2026-08-19:
@anrouxel, @MarcinOrlowski, and the @TheEvilSkeleton as maintainers of the Snap and Flatpak packages. This seems like a good time to revive this old open discussion. Do you see a way how we all can make our life easier by integrating building and publishing these packages automatically as part of our release process? @BFH-ktt1 and myself don't have experience with these package formats, but appreciate their usefulness to make Logisim-evolution accessible to a large user base. I have seen other projects, which README features badges listing the currently packaged versions on different platforms. This might be interesting, too.

TheEvilSkeleton, 2026-08-20:
Flathub has documentation for taking over existing packages: https://docs.flathub.org/docs/for-app-authors/submission#someone-else-has-put-my-app-on-flathubwhat-do-i-do

As for incorporating and automatically publishing to Flathub as part of the release process, I'm not too sure if they would allow that. The Flathub package lives in the [flathub/com.github.reds.LogisimEvolution](https://github.com/flathub/com.github.reds.LogisimEvolution) repository, which I have been updating from there ever since.

For badges, I use https://shields.io/badges/flathub-downloads for my personal projects.

maehne, 2026-09-07:
@MarcinOrlowski: Could you have a look at @hewzhew's analysis of the open steps to fix the Snap package publication again? I think that you should still have the rights to set up the necessary token.

maehne, 2026-09-07:
@TheEvilSkeleton: Thanks for reacting to this issue! We are thankful for your effort to maintain the Flatpak package and hope that you will find some time to fix the two open issues!

Thanks also for pointing us to [Shields.io](https://shields.io)! Augmenting our README.md with the version badges for the different reference package managers will be handy for our users to see what version of Logisim-evolution they can expect to get.