logisim-evolution/logisim-evolution #747: Firewall warning when launching Logisim-evolution

maehne, 2021-07-01:
I always thought that the firewall warning concerning allowing incoming network connections for Logisim-evolution was related to the Auto Updater, which has been removed in PR #650. However, I still receive this warning building from `HEAD` (commit 444b3ed1f18b92113f23cb5b8ed310665167154d) of the `develop` branch at least on macOS:

[image]

MarcinOrlowski, 2021-07-01:
I am completely ignorant when we talk macOS packages, but can it be that that this is rather related to the package not the app? Will the FW alert show if you run it manually from IDE or jar?

PS: also mind posting English version of this message?

MarcinOrlowski, 2021-07-13:
@davidhutchens could you please take look at this one?

maehne, 2021-07-13:
The firewall warning still appears after opening directly the freshly built app from `develop` using `open build/dist/Logisim-evolution.app` or the installed app, which I copy from the DMG file to the `Applications/` folder. The German warning message above translates to:

> **Do you want the programme "Logisim-evolution.app" to accept incoming network connections?**
> 
> Clicking on "Do not allow" may restrict the behaviour of the programme. This setting can be changed in the system setting "Security" in the section "Firewall".

@davidhutchens: Can you reproduce it on your side?

davidhutchens, 2021-07-13:
If the firewall is enabled, I get that message. I have normally clicked Deny and like @maehne assumed it was related to an update check. Checking Deny hasn't caused me any issues in running the program.

However, for me, either Allowing or Denying the connection sticks. It is recorded and can be seen in System Preferences under Security & Privacy, Firewall, Firewall Options. Even rebuilding the app after some changes didn't cause it to ask again. If I delete the line from Firewall Options, it asks on the next launch.

I am running macOS 10.15.7 with all security updates installed.

I'm not sure why it doesn't stick for you.

More importantly, according to Apple's documentation, https://support.apple.com/guide/mac-help/block-connections-to-your-mac-with-a-firewall-mh34041/mac, it only asks when there is an attempt to connect from outside. So why is there an attempt to connect from outside?

MarcinOrlowski, 2021-07-13:
> So why is there an attempt to connect from outside?

This: https://superuser.com/a/328802 is pretty old anwer but maybe can lead somewhere (also I see this is quite "popular" issue):

> Every app in OS X should be signed by the developer. This signature ensures that the contents of the application haven't been changed. The firewall verifies this signature as part of its process. When the signature verification process fails, you will receive this Allow/Deny error message every time you use the program, even if you click Allow on the dialog box.
> 
> Why is the signature verification failing? This can happen for several reasons, but primarily it is because somewhere along the line you've changed the contents of the /Applications/YourProblemApplication.app container. For example, if you use Monolingual or a similar app to remove other language files to save disk space, you've modified the contents of the app. If you've used a program like CandyBar or IconDubber to change the icon (.icns) file, you've also changed the contents of the app.

maehne, 2021-07-13:
I can confirm that also on my side, my choice for the Firewall settings gets saved. It is only shown again once I install a new Logisim-evolution version to the `/Applications/` folder. If signing the app would fix this issue, this is another reason to look into distributing a code-signed version on macOS.

davidhutchens, 2021-07-13:
We actually specifically unsign the app in the build (see last step in the createApp task in build.gradle.kts) because it was otherwise left in a state that macOS reported as damaged. We currently have no valid developer signature to use.

Yes, this problem is probably related to it being unsigned, and may involve how different versions of macOS handle it.

MarcinOrlowski, 2021-07-13:
Should we have "Troubleshoting" chapter added to our docs explaining this particular thing to end users?

maehne, 2021-07-14:
For the moment, I think such platform-specific particularities could be briefly mentioned in the Download section as a note.

davidhutchens, 2021-07-16:
I will compose something about platform-specific installation issues on macOS for the Download section. I will also fix the duplicate word @maehne noted in PR #819 and make another pass over the English in those sections.

MarcinOrlowski, 2021-07-16:
Great, thanks David.

maehne, 2021-07-22:
If we should decide to sign our macOS app at some point, there seems to be no way around paying the yearly 99 USD fee to obtain the [necessary Apple-signed certificate](https://developer.apple.com/programs/whats-included/). The good thing is that the certificate can be safely used also in GitHub Actions to automatically build and sign the application, as the [GitHub docs](https://docs.github.com/en/actions/guides/installing-an-apple-certificate-on-macos-runners-for-xcode-development) explain.