logisim-evolution/logisim-evolution #748: Check key handling code and reapply PRs #645 and #646

maehne, 2021-07-01:
@BFH-ktt1 I am not going to restore #645 and #646 for now as you raised the concern about possible conflict with other key handling code. Yet this feature is pretty handy to have, so can you please suggest better approach?

_Originally posted by @MarcinOrlowski in https://github.com/logisim-evolution/logisim-evolution/issues/680#issuecomment-862552275_

R3dst0ne, 2021-07-13:
I have no experienced with GUIs, but I might have two interesting points.
- In the documentation (`Guide to Being a Logisim User > Menue reference`) there are shortcuts described to close windows.
- The same code block gets defined for each window. Could it be moved to [gui/generic/LFrame.java](https://github.com/logisim-evolution/logisim-evolution/blob/develop/src/main/java/com/cburch/logisim/gui/generic/LFrame.java)?

maehne, 2026-09-28:
I agree with option 1 to close this issue as being addressed by PR #1802.

closed 2026-09-28 by maehne