logisim-evolution/logisim-evolution #749: Acknowledge the 3rd-party libraries we are using in Logisim-evolution

maehne, 2021-07-01:
With #719 this task is complete, other than point 1 in the comment by @BFH-ktt1. How should we acknowledge the maven libraries that we use? We could add them to the list of contributors in the README. We could also add them to the scroll in the About panel. Thoughts?

_Originally posted by @davidhutchens in https://github.com/logisim-evolution/logisim-evolution/issues/321#issuecomment-867587328_

maehne, 2021-07-01:
@davidhutchens and @BFH-ktt1: I agree that we should acknowledge the 3rd-party libraries we are using and I like the proposal to do so in the README and the About panel.

MarcinOrlowski, 2021-07-01:
I think that this one and #590 should be somehow combined into one effort.