logisim-evolution/logisim-evolution #753: Define contribution guidelines

maehne, 2021-07-01:
[...] it's high time to have processes defined. As for PRs I personally would enforce no direct pushes neither to master nor develop and at least one mandatory reviewer per PR. Shall PR be accompanied by related mandatory issue/FR ticket is another story - I'd say, yes (as discussion should take place in the ticket there, not in PR where one should discuss the PR itself). That would also help avoid situations where there's some work already done (like #722) and it's unclear what to do with it. Mandating  ticket first would save some efforts, esp. in case of questionable changes. [...]

_Originally posted by @MarcinOrlowski in https://github.com/logisim-evolution/logisim-evolution/issues/744#issuecomment-872101565_

MarcinOrlowski, 2021-07-02:
Another thing worth standardizing is how PR are merged. I think it should always be "squash" merge. We do loose details of works but in fact "who cares". PR should be kind of "atomic" - one thing at the time. So if it does not work properly, we revert just one commit and voila. Problem solved. I was working for major tech companies who merged that way and had never single issue over the years. It looks like there are benefits from having granular PR, but in reality it is just noise.

maehne, 2021-07-03:
I agree that squashing commits for PRs is in many cases a good idea. So, I'm fine to make it a recommendation. However, there are cases such as when refactoring code, where I like to keep the individual commits, which each tackle a different small issue.

MarcinOrlowski, 2026-09-05:
As we currently do have `.github/CONTRIBUTING.md` I am closing this ticket. Feel free to reopen if that file is not enogh (yet if that is so I'd recomment creating new ticket and specify what is still missing).

closed 2026-09-05 by MarcinOrlowski