logisim-evolution/logisim-evolution #756: Incorporate analytics (or similar tool)

MarcinOrlowski, 2021-07-01:
We got absolutely no idea about our users' habits. Can't tell what language version they use (from provided), what language they lack (theirs system language), can we axe i.e. animated icons or it's the best part of the Logisim (hint: no, it's not :). Not sure what are our alternatives, but we need somehow to gather more info as w/o that it's hard to plan anything. And of course, it must be opt-out, but that's rhetoric. I'd start with Google analytics (with perhaps some abstraction layer). Thoughts?

maehne, 2021-07-01:
I reckon that it would be helpful to know a bit more about our user base, but I am not sure your proposal is the right way. Personally, I am not a fan of opt-out analytics/telemetry functions in other software products. I systematically turn it off and know that some people even stopped using an application after it introduced analytics. If we should decide in the end to add something like that, then it should be very transparently communicated, opt-in and not opt-out, and preferably a solution, which does not require some of some big data collecting cloud provider. Maybe inviting our users to answer a questionnaire would be the better approach.

MarcinOrlowski, 2021-07-01:
I understand your reasoning but about "opt-in" vs "opt-out". But we need "opt-out" I am afraid or analytics bring us no data. We shall be open, announce introduction of analytics like one version prior, show the dialog on first run and ask etc. This should put us in pretty clear position - "we need to know that to deliver better product, **please** do not -opt-out if you use LS, but if you must, here's how." vs what most corpos are doing -> "to opt-out double click this hidden button with your left leg's 3rd finger, while standing on one leg, whistling" :) We must go opt-out as we must have this data. We may ask "hey, we need this data, please can you stay opt-in", but the defaults must rather be that opt-out is alternative, not them opt-in.

maehne, 2021-07-01:
Thinking further about this, I have the feeling that we are already in the fortunate position to be very close to our user base. @BFH-ktt1 and myself employ it in our digital design courses and many small and big improvements to Logisim-evolution are directly motivated by the feedback we get from our students when they employ this tool in larger design projects (running for 8 weeks). Another driving factor are our needs for teaching different topics in our courses. I think this is also the major driving force for other contributors to this repository and other Logisim forks.

Introducing such a controversial feature should be first discussed in broader round and there should be consensus in the end among all regular contributors for this approach or against it.

MarcinOrlowski, 2021-07-01:
What I am saying is that we need analytics-alike metrics. How to implement it "the right way" is the next step and I do agree it needs to be properly planned. And this is what this ticket is for :)

zdimension, 2021-07-28:
I absolutely agree that analytics is a must-have, but it's essential that we don't end up like Audacity when Muse Group added (opt-in) analytics a few weeks ago, which caused quite an uproar in the community